qmk/qmk_firmware #23537: [Bug] Caps Word get reset by normal press with interrupting characters on long press

JohannesFleischer, 2024-04-16:
### Describe the Bug

Caps Word gets reset when a non-English characters like the German `ö, ä , ü , ß` is pressed.
I understand that this is most likely by design to reduce the complexity, but this it even gets reset, if a key is like `a` is pressed normally, when it is configured to enter a `ä` on long press.

### Keyboard Used

zsa voyager

### Operating System

Ubuntu 23.04

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

getreuer, 2024-05-04:
Thanks for the report. How are you implementing umlaut on long press? (tap dance, or is this being done through Oryx?) You can configure which keys Caps Word considers "word breaking" with the `caps_word_press_user()` callback, see this page:

https://docs.qmk.fm/#/feature_caps_word?id=configure-which-keys-are-word-breaking

JohannesFleischer, 2024-05-04:
I used Oryx to configure my layout.
Thanks for pointing this out. I did not know this feature existed.
But it seems to be missing in Oryx so I will forward this to zsa then.

closed 2024-05-04 by JohannesFleischer