qmk/qmk_firmware #23549: Strange behaviour of macros with lily58

ckulemeyer, 2024-04-18:
### Describe the Bug

With these 3 Macros, one is always not working, depending on order or variation of macros. With the attached files, F_DEL returns an ordinary key:
  case B_DEL:
    if (record->event.pressed) {
      SEND_STRING(SS_LSFT(SS_TAP(X_HOME)) SS_TAP(X_DEL));
      /* tap_code(KC_DEL); */
    }
    return false;

  case F_DEL:
    if (record->event.pressed) {
      SEND_STRING(SS_LSFT(SS_TAP(X_END)) SS_TAP(X_DEL));
      /* tap_code(KC_DEL); */
    }
    return false;

  case L_DEL:
    if (record->event.pressed) {
      SEND_STRING(SS_TAP(X_HOME) SS_LSFT(SS_TAP(X_END)) SS_TAP(X_DEL));
      /* SEND_STRING(SS_TAP(X_DEL)); */
    }
    return false;

[lily58_keymap.zip](https://github.com/qmk/qmk_firmware/files/15021104/lily58_keymap.zip)

### Keyboard Used

lily58

### Operating System

windows/Ubuntu

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/vagrant/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.4 LTS).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.7
Ψ - Latest master: 2024-04-14 09:00:19 -0400 (9baafef968) -- Swap PID and VID on is0gr in QMK (#23458)
Ψ - Latest upstream/master: 2024-04-17 16:57:21 +0100 (aa15b64f55) -- [Keyboard] Keycult 65 (#23485)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-04-14 09:00:19 -0400 (9baafef968) -- Swap PID and VID on is0gr in QMK (#23458)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

ckulemeyer, 2024-04-20:
Sorry, I found the mistake. I just forgot a 'SAFE_RANGE'.

closed 2024-04-20 by ckulemeyer

PeterMortensen, 2024-06-22:
> Sorry, I found the mistake. I just forgot a 'SAFE_RANGE'.

Can you elaborate?

ckulemeyer, 2024-09-05:
sorry @PeterMortensen for the late response:
enum custom_keycodes {
  B_DEL = SAFE_RANGE,
  F_DEL,
  L_DEL,
  };