qmk/qmk_firmware #23592: [Bug] Compilation of TAP DANCE fails (code compiled previously).

robert-winkler, 2024-04-23:
### Describe the Bug

My hand-wired Dactyl Manuform has been working well for some years. Now, the keyboard started failing, especially in the heat, why I decided to change the mini-USB Arduino boards for new USB-C Arduinos.
I installed the current software and placed my keymap into the current git tree of the qmk_firmware. The keymap has several layers for multilingual typing with Colemak layout and a steno layer. Previously, it compiled without problems.
Now, it terminates with an error: 
~~~
In file included from quantum/keymap_introspection.c:5:0:
./keyboards/handwired/dactyl_manuform/4x6/keymaps/colemakdh_dactyl-manuform-4x6/keymap.c:455:12: error: ‘xtap_state’ defined but not used [-Werror=unused-variable]
 static tap xtap_state = {
            ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake: *** [builddefs/common_rules.mk:373: .build/obj_handwired_dactyl_manuform_4x6_colemakdh_dactyl-manuform-4x6/quantum/keymap_introspection.o] Error 1
~~~
You can see the complete code at:
[dactyl-keyboard/colemakdh_dactyl-manuform-4x6](https://github.com/robert-winkler/QMK/tree/main/dactyl-keyboard/colemakdh_dactyl-manuform-4x6)

The complete output of the compilation is given below.

### Keyboard Used

handwired/dactyl_manuform/4x6/keymaps/colemakdh_dactyl-manuform-4x6

### Link to product page (if applicable)

https://github.com/robert-winkler/QMK/tree/main/dactyl-keyboard

### Operating System

 Linux (Ubuntu 23.10)

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/rob/qmk_firmware
Ψ Detected Linux (Ubuntu 23.10).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.7
Ψ - Latest master: 2024-04-22 18:22:36 +0400 (e807bb7e37) -- Add Lisa keyboard (#23575)
Ψ - Latest upstream/master: 2024-04-22 18:22:36 +0400 (e807bb7e37) -- Add Lisa keyboard (#23575)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-04-22 18:22:36 +0400 (e807bb7e37) -- Add Lisa keyboard (#23575)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed
[... 146 lines cut ...]
./keyboards/handwired/dactyl_manuform/4x6/keymaps/colemakdh_dactyl-manuform-4x6/keymap.c:486:62: note: (near initialization for ‘tap_dance_actions[0]’)
     [X_CTL] = ACTION_TAP_DANCE_FN_ADVANCED(NULL, x_finished, x_reset)
                                                              ^
quantum/process_keycode/process_tap_dance.h:73:62: note: in definition of macro ‘ACTION_TAP_DANCE_FN_ADVANCED’
     { .fn = {user_fn_on_each_tap, user_fn_on_dance_finished, user_fn_on_dance_reset, NULL}, .user_data = NULL, }
                                                              ^
quantum/process_keycode/process_tap_dance.h:73:93: error: field name not in record or union initializer
     { .fn = {user_fn_on_each_tap, user_fn_on_dance_finished, user_fn_on_dance_reset, NULL}, .user_data = NULL, }
                                                                                             ^
./keyboards/handwired/dactyl_manuform/4x6/keymaps/colemakdh_dactyl-manuform-4x6/keymap.c:486:15: note: in expansion of macro ‘ACTION_TAP_DANCE_FN_ADVANCED’
     [X_CTL] = ACTION_TAP_DANCE_FN_ADVANCED(NULL, x_finished, x_reset)
               ^
quantum/process_keycode/process_tap_dance.h:73:93: note: (near initialization for ‘tap_dance_actions[0]’)
     { .fn = {user_fn_on_each_tap, user_fn_on_dance_finished, user_fn_on_dance_reset, NULL}, .user_data = NULL, }
                                                                                             ^
./keyboards/handwired/dactyl_manuform/4x6/keymaps/colemakdh_dactyl-manuform-4x6/keymap.c:486:15: note: in expansion of macro ‘ACTION_TAP_DANCE_FN_ADVANCED’
     [X_CTL] = ACTION_TAP_DANCE_FN_ADVANCED(NULL, x_finished, x_reset)
               ^
In file included from quantum/keymap_introspection.c:5:0:
./keyboards/handwired/dactyl_manuform/4x6/keymaps/colemakdh_dactyl-manuform-4x6/keymap.c:455:12: error: ‘xtap_state’ defined but not used [-Werror=unused-variable]
 static tap xtap_state = {
            ^
cc1: all warnings being treated as errors
gmake: *** [builddefs/common_rules.mk:373: .build/obj_handwired_dactyl_manuform_4x6_colemakdh_dactyl-manuform-4x6/quantum/keymap_introspection.o] Error 1
~~~

zvecr, 2024-04-23:
This isnt a bug in QMK. The various keymap errors stem from various parts not being up to date.

Tap dance:
> error: unknown type name ‘qk_tap_dance_state_t’
 uint8_t cur_dance(qk_tap_dance_state_t *state);

https://github.com/qmk/qmk_firmware/pull/19313 from the 2023 Q1 breaking changes.

Encoder:
> error: conflicting types for ‘encoder_update_user’
 void encoder_update_user(uint8_t index, bool clockwise) {

https://docs.qmk.fm/#/ChangeLog/20210529?id=encoder-callback-boolean from the 2021 Q2 breaking changes.

Leader:

> error: data definition has no type or storage class [-Werror]
 LEADER_EXTERNS();

https://docs.qmk.fm/#/ChangeLog/20230226?id=leader-key-rework from the 2023 Q1 breaking changes.

closed 2024-04-23 by zvecr

robert-winkler, 2024-04-23:
Amazing! Thanks a lot, @zvecr for the quick and competent support !!! 

Anyway, it would be cool to have a summary somewhere on syntax changes. Or even a script similar to the (in)famous Python `2to3.py`  😉.