qmk/qmk_firmware #23619: [Bug] Idobao ID75 V3 (Drop) Issue with RGB Matrix

DutchMex, 2024-04-26:
### Describe the Bug

When using the QMK firmware for Idobao id75/v2 on my ID75 v3 keyboard from Drop, there are glitches when using any of the preconfigured RGB animations. Particularly, any moving or reactive light effect, when nearing the top row, will cause many other LEDs in the matrix to flash with rainbow colors. It seems as though the lighting overflows what is physically present on the keyboard, and then the overflow causes “random” flashes of color on other LEDs. Sometimes this will even result in a random few LEDs remaining on after turning off RGB completely.

To reproduce bug:
- Flash ID75 V3 with ID75 V2 QMK profile from QMK Configurator.
- Turn on RGB effects
- Select a reactive effect and begin typing near the top or sides of the keyboard.
- The lighting should start to flash and glitch

### Keyboard Used

id75/v2

### Link to product page (if applicable)

https://drop.com/buy/idobao-id75-v3-hot-swappable-ortholinear-aluminum-keyboard-kit

### Operating System

Mac OS Sonoma 14.4.1

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

N/A

### Additional Context

I have very little experience with QMK (other than QMK configurator and QMK-Toolbox), but a good bit of experience with microcontrollers and code, so if anyone is willing to assist, I would love to work on a solution to this. Thanks so much!

drashna, 2024-05-28:
> When using the QMK firmware for Idobao id75/v2 on my ID75 v3 keyboard from Drop

Are these actually the same hardware layout? 

And are there 7 leds that flash like that?

DutchMex, 2024-05-29:
I’m not sure if they are or not. I would suspect that it is at least slightly different, since Drop referred to it as V3.

Unsure about 7 leds, since it is my main keyboard at work and haven’t had time to take it apart to count them. It is exactly the top three rows of per-key LEDs though, and I don’t know whether the backlight ones are doing the same thing or not, since this is a solid metal case, and I have not had time to take it apart recently, as noted previously.

So, like 45 per-key LEDs that misbehave as described above, and unknown behavior for under keyboard LEDs.

egormanga, 2025-11-15:
There are total 14 underglow leds, and the current `ymdk/id75/f103` definition seems to confuse their physical order for the bottom row:
```patch
--- a/keyboards/ymdk/id75/info.json
+++ b/keyboards/ymdk/id75/info.json
@@ -134,13 +134,13 @@
             {"matrix": [0, 2], "x": 32, "y": 0, "flags": 4},
             {"matrix": [0, 1], "x": 16, "y": 0, "flags": 4},
             {"matrix": [0, 0], "x": 0, "y": 0, "flags": 4},
-            {"x": 0, "y": 64, "flags": 2},
-            {"x": 38, "y": 64, "flags": 2},
-            {"x": 76, "y": 64, "flags": 2},
-            {"x": 114, "y": 64, "flags": 2},
-            {"x": 152, "y": 64, "flags": 2},
-            {"x": 190, "y": 64, "flags": 2},
             {"x": 224, "y": 64, "flags": 2},
+            {"x": 190, "y": 64, "flags": 2},
+            {"x": 152, "y": 64, "flags": 2},
+            {"x": 114, "y": 64, "flags": 2},
+            {"x": 76, "y": 64, "flags": 2},
+            {"x": 38, "y": 64, "flags": 2},
+            {"x": 0, "y": 64, "flags": 2},
             {"x": 0, "y": 0, "flags": 2},
             {"x": 38, "y": 0, "flags": 2},
             {"x": 76, "y": 0, "flags": 2},
```
I don't know about the `idobao/id75/v2` definition, nor if it should be used with this model.