qmk/qmk_firmware #23654: [Bug] xd75 doesn't build on head

xster, 2024-05-03:
### Describe the Bug

- `brew upgrade qmk`
- sync qmk_firmware to tip of tree
- `qmk compile -kb xd75 -km default`

```
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=xiudi/xd75 KEYMAP=default KEYBOARD_FILESAFE=xiudi_xd75 TARGET=xiudi_xd75_default INTERMEDIATE_OUTPUT=.build/obj_xiudi_xd75_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

Generating: .build/obj_xiudi_xd75_default/src/info_deps.d                                           [OK]
avr-gcc (Homebrew AVR GCC 8.5.0_2) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_xiudi_xd75_default/src/info_config.h                                         [OK]
Generating: .build/obj_xiudi_xd75_default/src/default_keyboard.c                                    [OK]
Generating: .build/obj_xiudi_xd75_default/src/default_keyboard.h                                    [OK]
Compiling: keyboards/xiudi/xd75/xd75.c                                                              [OK]
Compiling: .build/obj_xiudi_xd75_default/src/default_keyboard.c                                     [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/main.c                                                                           [OK]
Assembling: platforms/avr/xprintf.S                                                                 [OK]
Compiling: platforms/avr/printf.c                                                                   [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight/rgblight.c                                                              [OK]
Compiling: quantum/rgblight/rgblight_drivers.c                                                      [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: quantum/backlight/backlight.c                                                            [OK]
Compiling: quantum/process_keycode/process_backlight.c                                              [OK]
Compiling: quantum/backlight/backlight_driver_common.c                                              [OK]
Compiling: platforms/avr/drivers/backlight_pwm.c                                                   platforms/avr/drivers/backlight_pwm.c: In function 'enable_pwm':
platforms/avr/drivers/backlight_pwm.c:133:5: error: 'TCCRxA' undeclared (first use in this function); did you mean 'TCCR0A'?
     TCCRxA |= _BV(COMxx1) | _BV(COMxx0);
     ^~~~~~
     TCCR0A
platforms/avr/drivers/backlight_pwm.c:133:5: note: each undeclared identifier is reported only once for each function it appears in
In file included from /opt/homebrew/Cellar/avr-gcc@8/8.5.0_2/avr/include/avr/io.h:99,
                 from platforms/avr/_pin_defs.h:18,
                 from platforms/pin_defs.h:22,
                 from platforms/gpio.h:18,
                 from platforms/avr/drivers/backlight_pwm.c:2:
platforms/avr/drivers/backlight_pwm.c:133:19: error: 'COMxx1' undeclared (first use in this function); did you mean 'COM1B1'?
     TCCRxA |= _BV(COMxx1) | _BV(COMxx0);
                   ^~~~~~
platforms/avr/drivers/backlight_pwm.c:133:33: error: 'COMxx0' undeclared (first use in this function); did you mean 'COM0A0'?
     TCCRxA |= _BV(COMxx1) | _BV(COMxx0);
                                 ^~~~~~
platforms/avr/drivers/backlight_pwm.c: In function 'disable_pwm':
platforms/avr/drivers/backlight_pwm.c:141:5: error: 'TCCRxA' undeclared (first use in this function); did you mean 'TCCR0A'?
     TCCRxA &= ~(_BV(COMxx1) | _BV(COMxx0));
     ^~~~~~
     TCCR0A
[... 78 lines cut ...]
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

xster, 2024-05-03:
oops my bad. I didn't in fact merge correctly with tip of tree. Thought I did yesterday :S 

`qmk compile -kb xd75 -km default` does indeed build on qmk/qmk_firmware:master. Sorry for the noise.

xster, 2024-05-03:
Perhaps what I should be really asking is that there seems to be a regression on xd75 on head with the backlight feature. 

Putting aside my userspace keymap for the time being, setting https://github.com/qmk/qmk_firmware/blob/8075003e6068d6f8fa5396972c2a17d39d14c584/keyboards/xiudi/xd75/rules.mk#L10 to yes no longer seems to work. But the hardware itself does have LEDs on the board and a per key LED which used to work in the past.

zvecr, 2024-05-03:
The following compiles fine on develop
```diff
diff --git a/keyboards/xiudi/xd75/keyboard.json b/keyboards/xiudi/xd75/keyboard.json
index a928b43f9b..7d0e6c1bfc 100644
--- a/keyboards/xiudi/xd75/keyboard.json
+++ b/keyboards/xiudi/xd75/keyboard.json
@@ -15,6 +15,7 @@
         "extrakey": true,
         "mousekey": true,
         "nkro": false,
+        "backlight": true,
         "rgblight": true
     },
     "matrix_pins": {
@@ -24,6 +25,7 @@
     "diode_direction": "COL2ROW",
     "backlight": {
         "pin": "F5",
+        "driver": "timer",
         "levels": 6,
         "on_state": 0
     },
```

xster, 2024-05-03:
I see. I suppose it works on master too with 

```
BACKLIGHT_ENABLE = yes
BACKLIGHT_DRIVER = timer
```

Unclear how users would know to set BACKLIGHT_DRIVER as well. Perhaps this was implicit at one point further up in the tree when rules.mk for this keyboard used to set the MCU, etc. Not sure if related. 

Either way, I can build my keymap on master now. Thanks!

closed 2024-05-03 by xster