qmk/qmk_firmware #23721: Add issue template for documentation requests (and hi, I'm back!)

NoahAndrews, 2024-05-15:
### Issue Description

We should add templates for problems with existing documentation and requests for new documentation.

---

Also, hey! I used to be a regular contributor to the QMK docs, [way back in 2016](https://github.com/qmk/qmk_firmware/commits?author=NoahAndrews). All this time I haven't had a keyboard that runs QMK, but I'm finally working on getting my new Kyria up and running.

It's amazing to see how much progress has been made in the last 7 years, most definitely including the docs. Still, I'm already finding things here and there that I'd like to improve. No guarantees I'll become a regular contributor again, but I could see it :)

NoahAndrews, 2024-08-14:
Still relevant, I think

tzarc, 2024-11-13:
No action by OP, closing as per comment on https://github.com/qmk/qmk_firmware/issues/23722#issuecomment-2290526113.

closed 2024-11-13 by tzarc