qmk/qmk_firmware #23736: [Bug] "make sofle:rgb_default" fails with "RGBLED_SPLIT" redefined error

mdzyuba, 2024-05-18:
### Describe the Bug

The make sofle:rgb_default fails:

```
$ make sofle:rgb_default
QMK Firmware 0.24.9
Making sofle/rev1 with keymap rgb_default

Generating: .build/obj_sofle_rev1_rgb_default/src/info_deps.d                                       [OK]
avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_sofle_rev1_rgb_default/src/info_config.h                                     [OK]
Generating: .build/obj_sofle_rev1_rgb_default/src/default_keyboard.c                                [OK]
Generating: .build/obj_sofle_rev1_rgb_default/src/default_keyboard.h                                [OK]
Compiling: keyboards/sofle/sofle.c                                                                 In file included from <command-line>:0:0:
./keyboards/sofle/keymaps/rgb_default/config.h:73:0: error: "RGBLED_SPLIT" redefined [-Werror]
  #define RGBLED_SPLIT { 35, 35 } // haven't figured out how to use this yet
 ^
In file included from <command-line>:0:0:
./.build/obj_sofle_rev1_rgb_default/src/info_config.h:45:0: note: this is the location of the previous definition
 #    define RGBLED_SPLIT { 36, 36 }
 ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [builddefs/common_rules.mk:373: .build/obj_sofle_rev1_rgb_default/keyboards/sofle/sofle.o] Error 1
Make finished with errors
make: *** [Makefile:415: sofle:rgb_default] Error 1
```

### Keyboard Used

sofle

### Operating System

Linux

### qmk doctor Output

```text
$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/mykola/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.4 LTS).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.9
Ψ - Latest master: 2024-05-13 20:20:47 +0200 (fcbbaf4485) -- Fix for RGB color override and brightness for EC Type K (#23703)
Ψ - Latest upstream/master: 2024-05-17 22:59:45 +0100 (47bc02b2ff) -- Force CPI update using timer when using split pointing. (#23545)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-05-13 20:20:47 +0200 (fcbbaf4485) -- Fix for RGB color override and brightness for EC Type K (#23703)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

aksdad, 2024-05-20:
Commenting or removing that should help you compile for now.
Since its already defined in the info.json https://github.com/qmk/qmk_firmware/blob/2fd56317763e8b3b73f0db7488ef42a70f5b946e/keyboards/sofle/info.json#L40

drashna, 2024-05-28:
I would recommend against using the rgb_default keymap, as it appears to have a bunch of  extra stuff that shouldn't be needed, and is definitely causing issues.