qmk/qmk_firmware #23775: [Bug]

rynfox13, 2024-05-22:
### Describe the Bug

There's a bug in the RGB code causing a compile time issue:

```
❯ qmk compile
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=handwired/dactyl_manuform/5x6 KEYMAP=kaze KEYBOARD_FILESAFE=handwired_dactyl_manuform_5x6 TARGET=handwired_dactyl_manuform_5x6_kaze INTERMEDIATE_OUTPUT=.build/obj_handwired_dactyl_manuform_5x6_kaze VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

avr-gcc (Homebrew AVR GCC 8.5.0_2) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  22382	      0	  22382	   576e	handwired_dactyl_manuform_5x6_kaze.hex

Compiling: .build/obj_handwired_dactyl_manuform_5x6_kaze/src/default_keyboard.c                     [OK]
Compiling: quantum/keymap_introspection.c                                                           [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/split_common/split_util.c                                                        [OK]
Compiling: quantum/split_common/transport.c                                                         [OK]
Compiling: quantum/split_common/transactions.c                                                      [OK]
Compiling: quantum/main.c                                                                           [OK]
Assembling: platforms/avr/xprintf.S                                                                 [OK]
Compiling: platforms/avr/printf.c                                                                   [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight/rgblight.c                                                             In file included from quantum/rgblight/rgblight.c:21:
quantum/rgblight/rgblight.h:21: error: "RGBLIGHT_LED_COUNT" redefined [-Werror]
 #    define RGBLIGHT_LED_COUNT RGBLED_NUM

In file included from <command-line>:
./.build/obj_handwired_dactyl_manuform_5x6_kaze/src/info_config.h:61: note: this is the location of the previous definition
 #    define RGBLIGHT_LED_COUNT 12

cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
gmake: *** [builddefs/common_rules.mk:373: .build/obj_handwired_dactyl_manuform_5x6_kaze/quantum/rgblight/rgblight.o] Error 1
```

Very simple fix, add a `#undef` before the `#define` (see additional detail below).

### Keyboard Used

handwire/dactyl_manuform/5x6

### Operating System

macOS

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Patch:

```patch
diff --git a/quantum/rgblight/rgblight.h b/quantum/rgblight/rgblight.h
index 9e2b073776..4b02e8b63f 100644
--- a/quantum/rgblight/rgblight.h
+++ b/quantum/rgblight/rgblight.h
@@ -18,6 +18,7 @@

 // DEPRECATED DEFINES - DO NOT USE
 #if defined(RGBLED_NUM)
+#    undef RGBLIGHT_LED_COUNT
 #    define RGBLIGHT_LED_COUNT RGBLED_NUM
 #endif
 // ========
```

drashna, 2024-05-28:
update your repo.  I can't reproduce this with the default keymap, and enabling rgblight from the command line.  

Also, run `qmk clean -a` and retry.

medoix, 2024-06-23:
I too have just stumbled across this when attempting to build the ```sofle/rev1``` keyboard using the ```rgb_default``` keymap. I have updated to latest master and comeplted a ```qmk clean -a``` but still get the same error.

```
[medoix@PC qmk_firmware]$ qmk compile -kb sofle/rev1 -km rgb_default                                                                              
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=sofle/rev1 KEYMAP=rgb_default KEYBOARD_FILESAFE=sofle_rev1 TARGET=sofle
INTERMEDIATE_OUTPUT=.build/obj_sofle_rev1_rgb_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"                                            
                                                                                                                                                     
                                                                                                                                                     
Generating: .build/obj_sofle_rev1_rgb_default/src/info_deps.d                                       [OK]                                             
avr-gcc.exe (GCC) 8.5.0                                                                                                                              
Copyright (C) 2018 Free Software Foundation, Inc.                                                                                                    
This is free software; see the source for copying conditions.  There is NO                                                                           
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.                                                                          
                                                                                                                                                     
Generating: .build/obj_sofle_rev1_rgb_default/src/info_config.h                                     [OK]                                             
Generating: .build/obj_sofle_rev1_rgb_default/src/default_keyboard.c                                [OK]                                             
Generating: .build/obj_sofle_rev1_rgb_default/src/default_keyboard.h                                [OK]                                             
Compiling: keyboards/sofle/sofle.c                                                                 In file included from <command-line>:             
./keyboards/sofle/keymaps/rgb_default/config.h:73: error: "RGBLED_SPLIT" redefined [-Werror]                                                         
  #define RGBLED_SPLIT { 35, 35 } // haven't figured out how to use this yet                                                                         
                                                                                                                                                     
In file included from <command-line>:                                                                                                                
./.build/obj_sofle_rev1_rgb_default/src/info_config.h:45: note: this is the location of the previous definition                                      
 #    define RGBLED_SPLIT { 36, 36 }                                                                                                                 
                                                                                                                                                     
cc1.exe: all warnings being treated as errors                                                                                                        
 [ERRORS]                                                                                                                                            
 |                                                                                                                                                   
 |                                                                                                                                                   
 |                                                                                                                                                   
make: *** [builddefs/common_rules.mk:373: .build/obj_sofle_rev1_rgb_default/keyboards/sofle/sofle.o] Error 1
```

QMK Doctor
```
[medoix@PC qmk_firmware]$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: C:/Users/medoix/qmk_firmware
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.7.2
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.25.8
Ψ - Latest master: 2024-06-23 11:33:01 +1000 (64a579b1a8) -- Bump submodules
Ψ - Latest upstream/master: 2024-06-22 05:45:04 +0100 (7aa2ce2b38) -- Update documentation suggestion in top level readme (#23978)
Ψ - Latest upstream/develop: 2024-06-22 12:14:17 +0100 (7824e7ed9b) -- Migrate `led_update_kb` implementations to DD (#23983)
Ψ - Common ancestor with upstream/master: 2024-06-22 05:45:04 +0100 (7aa2ce2b38) -- Update documentation suggestion in top level readme (#23978)
Ψ - Common ancestor with upstream/develop: 2024-06-22 05:45:04 +0100 (7aa2ce2b38) -- Update documentation suggestion in top level readme (#23978)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 0.7.2
[... cut ...]
Ψ Submodule status:
Ψ - lib/chibios: 2023-01-03 19:29:26 +0000 --  (0062927e3)
Ψ - lib/chibios-contrib: 2023-01-11 16:42:27 +0100 --  (a224be15)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

zvecr, 2026-09-20:
`RGBLED_NUM` has been deprecated for over 3 years, and backwards compatibility (be it still broken) is being removed this cycle.

closed 2026-09-20 by zvecr