qmk/qmk_firmware #23798: [Bug] Compile of keebio/iris/rev8 fails

rdmi4, 2024-05-26:
### Describe the Bug

"qmk compile -kb keebio/iris/rev8 -km default" fails with many warnings / errors, the first few of which follow.

Compiling: lib/pico-sdk/src/rp2_common/hardware_clocks/clocks.c                                    /tmp/ccr4bW1F.s: Assembler messages:
/tmp/ccr4bW1F.s:219: Error: cannot honor width suffix -- `lsl r2,r2,#5'
/tmp/ccr4bW1F.s:224: Error: cannot honor width suffix -- `eor r2,r3'
/tmp/ccr4bW1F.s:226: Error: cannot honor width suffix -- `mov r3,#224'
/tmp/ccr4bW1F.s:227: Error: cannot honor width suffix -- `and r2,r3'

Looks like the ARM cross-compiler is generating code that is incompatible with the cross-assembler. Any suggestions?

### Keyboard Used

keebio/iris/rev8

### Operating System

OpenSUSE Tumbleweed

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /fserver/homeserver/dickrp/homedir/default/qmk_firmware
Ψ Detected Linux (openSUSE Tumbleweed).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.24.11
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2024-05-21 19:49:49 -0700 (7620c64b99) -- Added MATRIX_HAS_GHOST definition for IBM Model H controller (#23744)
Ψ - Latest upstream/master: 2024-05-21 19:49:49 -0700 (7620c64b99) -- Added MATRIX_HAS_GHOST definition for IBM Model H controller (#23744)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-05-21 19:49:49 -0700 (7620c64b99) -- Added MATRIX_HAS_GHOST definition for IBM Model H controller (#23744)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 4.9.3
Ψ Found avr-gcc version 3.4.4
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb1610)
Ψ - lib/chibios-contrib: 2023-11-27 18:15:44 +0100 --  (9d7a7f90)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-05-26:
Upgrade the compiler. 4.9.3 is almost 9 years old.

rdmi4, 2024-05-26:
Thank you! It builds now. I forgot I had that one installed.

closed 2024-05-26 by rdmi4