qmk/qmk_firmware #23801: Reduce number of HID interface descriptors?

jjlee, 2024-05-26:
### Issue Description

Is it possible to configure some keyboard USB interfaces to not be HID?  I'm using the hotdox driver, and currently in lsusb I see three interfaces, one of which has two endpoints, the rest only one endpoint, so four HID endpoints in total.

I'm having trouble attaching a hotdox to a device that doesn't like keyboards to have too many USB endpoints: keystrokes get lost.  Plugged in elsewhere, it works fine.  I'm hoping that for the purposes of this picky device, "endpoints" here turns out to mean "HID endpoints" so that if I can set some interfaces to not be identified as HID at all, the device will be happy.

What I've tried: I only use the keyboard for simple typing, so I tried configured `MOUSEKEY_ENABLE` and `EXTRAKEY_ENABLE` to `no` in `keyboards/hotdox/rules.mk` in the hope that might result in fewer HID interfaces, but I still see all three interfaces marked as `InterfaceClass         3 Human Interface Device`.

If this isn't currently possible via configuration, I'd appreciate any pointers to relevant parts of the code where I might begin to change this!

sigprof, 2024-05-26:
To disable the so called “shared endpoint”, you need to disable all of its users — in addition to `MOUSEKEY_ENABLE` and `EXTRAKEY_ENABLE`, there is also `NKRO_ENABLE`, which is also enabled in the `hotdox` config.  And the interface with two endpoints is probably the console interface, which is also enabled in the `hotdox` config; you may use `CONSOLE_ENABLE = no` to override that.

Also you should set all those options in a `rules.mk` file in your personal keymap folder instead of editing the existing `rules.mk` file for the keyboard.

And if your problematic device has a proper HID parser, or uses just the boot protocol for the keyboard, you may try setting `KEYBOARD_SHARED_EP = yes` instead of disabling the mousekey/extrakey/nkro options.

jjlee, 2024-05-26:
Setting NKRO_ENABLE and CONSOLE_ENABLE both to `no` too worked, thanks!

closed 2024-05-26 by jjlee