qmk/qmk_firmware #23824: [Bug] Multiple outputs from same column on press of 3 key (3wesd), Viktus Styrka MX STM32F072

nopunin10did, 2024-05-29:
### Describe the Bug

**Keyboard**
Viktus Styrka (MX version)

**Firmware where Bug Occurs**
1. Default firmware compiled by QMK Configurator ([default styrka](https://config.qmk.fm/#/viktus/styrka/LAYOUT_all))
2. Personal configuration compiled by QMK Configurator (see attached JSON)
3. Default firmware compiled locally using a recent pull of the QMK master branch
4. Personal configuration compiled locally using a recent pull of the QMK master branch (see attached JSON)

[viktus_styrka_nopunin10did_20240529.json](https://github.com/qmk/qmk_firmware/files/15486654/viktus_styrka_nopunin10did_20240529.json)

**Description of Problem**
When the 3 key is pressed (row 1, column 2 on the matrix), approximately 5%-10% of the time there appears to be an overflow of further characters. These characters appear as `32wesd` on the default layout and sometimes `32wesd\`` on my personal layout (since I map the left alt key to operate as a backtick/tilde key when tapped). Using QMK configurator's test page, I can confirm that the left Alt key is being read as pressed as well, along with the `32wesd` characters.

These appear to be keys in column 2 of the key matrix. I am not sure why the remaining keys on the fourth physical row of the keyboard are not also output (`zx` for the default firmware, `xc` on my personal firmware).

I have verified that this is not a soldering / hardware issue. The bug appears to occur most often shortly after the keyboard has been flashed.

**Possible Related Bug**
[Anne Pro 2 (C15) Nonfunctional #19886](https://github.com/qmk/qmk_firmware/issues/19886)

In the bug linked above, the reporter found what might have been a matrix scanning issue. However, the Anne Pro 2 uses an entirely different MCU, so this may just be coincidence.

### Keyboard Used

viktus/styrka

### Link to product page (if applicable)

https://viktus.design/collections/all/products/styrka

### Operating System

Windows 10 Pro

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.0
Ψ QMK home: C:/Users/alex.ronke/qmk_firmware
Ψ Detected Windows 10 (10.0.19045).
Ψ QMK MSYS version: 1.7.1
Ψ Userspace enabled: False
Ψ Git branch: styrka
Ψ Repo version: 0.25.1
Ψ - Latest styrka: 2024-05-29 02:54:47 +0100 (ef934355f9) -- Workaround for broken ChibiOS startup (#23822)
Ψ - Latest upstream/master: 2024-05-29 02:54:47 +0100 (2f9f000d0c) -- Workaround for broken ChibiOS startup (#23822)
Ψ - Latest upstream/develop: 2024-05-29 06:53:48 +0200 (4d32073681) -- BastardKB: remove legacy board `tbk` (#23818)
Ψ - Common ancestor with upstream/master: 2024-05-29 02:54:47 +0100 (2f9f000d0c) -- Workaround for broken ChibiOS startup (#23822)
Ψ - Common ancestor with upstream/develop: 2024-05-29 02:54:47 +0100 (2f9f000d0c) -- Workaround for broken ChibiOS startup (#23822)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 6.4
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305f)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

nopunin10did, 2024-06-13:
After recompiling and reflashing, I'm still getting this issue, but with a different key.

The 4 key yields:
45rtfg

Again I think these are all in one column of the matrix.

tzarc, 2024-06-13:
Try adding something like this to your keymap's `config.h` to see if it helps:
```c
#define MATRIX_IO_DELAY 100
```

Or, seeing as you've got a json:
```json
{
	/* the rest of your keymap here */
	"config": {
		"matrix_pins": {
			"io_delay": 100
		}
	}
}