qmk/qmk_firmware #23825: [Bug] Keyboard works when connected to laptop, but not through dock (modelh, thinkpad thunderbolt 3 workstation dock)

n-kremeris, 2024-05-29:
### Describe the Bug

I'm using an IBM Model M keyboard, upgraded with the [modelh controller](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ibm/model_m/modelh) from keebd.

The keyboard works perfectly fine when attached directly to any of my 3 laptops (archlinux thinkpad p1g2, windows11 thinkpad x1 carbon gen 8, windows11 t490s) but does not work when attached through my thinkpad thunderbolt 3 workstation dock gen2.

Attaching the keyboard directly to the laptop results in the following dmesg output:
`
[10388.938531] usb 1-2: new full-speed USB device number 25 using xhci_hcd
[10389.081163] usb 1-2: New USB device found, idVendor=1eaf, idProduct=0003, bcdDevice= 2.01
[10389.081166] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10389.081168] usb 1-2: Product: Maple 003
[10389.081168] usb 1-2: Manufacturer: LeafLabs
[10389.081169] usb 1-2: SerialNumber: LLM 003
[10390.451680] usb 1-2: USB disconnect, device number 25
[10390.775181] usb 1-2: new full-speed USB device number 26 using xhci_hcd
[10390.916535] usb 1-2: New USB device found, idVendor=feed, idProduct=b155, bcdDevice= 1.00
[10390.916539] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[10390.916541] usb 1-2: Product: Model M
[10390.916542] usb 1-2: Manufacturer: IBM
[10390.918142] input: IBM Model M as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:FEED:B155.0019/input/input47
[10390.972071] hid-generic 0003:FEED:B155.0019: input,hidraw2: USB HID v1.11 Keyboard [IBM Model M] on usb-0000:00:14.0-2/input0
[10390.973047] input: IBM Model M System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:FEED:B155.001A/input/input48
[10391.028550] input: IBM Model M Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:FEED:B155.001A/input/input49
[10391.028630] hid-generic 0003:FEED:B155.001A: input,hidraw3: USB HID v1.11 Device [IBM Model M] on usb-0000:00:14.0-2/input1
[10391.571392] audit: type=1326 audit(1717014109.109:20): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=22467 comm="qmk_id" exe="/usr/lib/udev/qmk_id" sig=31 arch=c000003e syscall=67 compat=0 ip=0x79b44dbec1ab code=0x80000000
[10391.572936] audit: type=1326 audit(1717014109.113:21): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=22473 comm="qmk_id" exe="/usr/lib/udev/qmk_id" sig=31 arch=c000003e syscall=67 compat=0 ip=0x79ef376c01ab code=0x80000000

`

And connecting the keyboard to the dock results in the following:
`
[10415.288333] usb 3-2.1.3: new low-speed USB device number 31 using xhci_hcd
[10415.415233] usb 3-2.1.3: device descriptor read/64, error -32
[10415.648606] usb 3-2.1.3: device descriptor read/64, error -32
[10415.878582] usb 3-2.1.3: new low-speed USB device number 32 using xhci_hcd
[10416.004928] usb 3-2.1.3: device descriptor read/64, error -32
[10416.238404] usb 3-2.1.3: device descriptor read/64, error -32
[10416.345138] usb 3-2.1-port3: attempt power cycle
`

The dock has the latest firmware installed, and I have built and flashed the latest version of QMK to the controller, resulting in no change so far. I have also tried disabling NKRO and mouse keys by adding a rules.mk file in the relevant directory, but make reported that both were already disabled in the jsonfile, so nothing changed from that either.

I'm hoping it's possible i could tweak something in the firmware that may allow the dock to recognize this controller, and would really appreciate some pointers - this is my first QMK based controller, and internet search results including "qmk" and "dock" are very few.

### Keyboard Used

ibm/model_m/modelh

### Link to product page (if applicable)

https://keebd.com/products/model-h-usb-controller-upgrade-ibm-model-m

### Operating System

archlinux, windows11

### qmk doctor Output

```text
not relevant
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2024-05-29:
Are you using a different cable to connect your keyboard to the dock?  Usually a misdetection of a full speed USB device as low speed happens when the D+ and D- connections are swapped for some reason.

n-kremeris, 2024-05-29:
No, i have tried 2 different cables, both straight to laptop and through the dock, same behavior persists

sigprof, 2024-05-30:
Looks like the problem happens even before the firmware is started (on STM32F103 the bootloader starts before the main firmware, and your board apparently has one of the stm32duino versions which shows up as the bootloader on USB initially).

It's really strange, because the USB circuit on the controller includes a permanent 1.5kΩ pull-up on D+, which should cause the USB host to detect at least a full speed USB device (there is also a separate procedure for detecting a high speed device, which does not apply here).  The same pull-up attached to D- should result in a detection of a low speed device; however, a weak pull-up in the MCU (≈50kΩ) should not have the same effect (the host side pull-down should be 15kΩ, which would result in about 0.76V on the data line, which is still below the V<sub>IL</sub> of 0.8V specified for the receiver).  Although maybe there is some weird spec violation somewhere, or some transient which causes the USB contoller in the dock to misdetect the device speed.

Can you try to add an extra USB hub between the dock and the keyboard?

Another thing that you could try is to add the following `board.h` file to your custom keymap:
```c
#pragma once

#include_next <board.h>

#undef VAL_GPIOAODR
#define VAL_GPIOAODR 0xFFFFF7FF
```
This will configure the A11 pin (D-) to use a pull-down instead of a pull-up, and also remove a potential high level output glitch on that pin (the GPIO init code writes ODR before CRH and CRL).

Make sure to run `qmk clean` after adding the `board.h` file, otherwise some parts of the code may not be rebuilt properly.

n-kremeris, 2024-05-30:
Hello again,

I will borrow a spare usb hub from work to test your suggestion next week.

Regarding the addition of board.h and pullup to pulldown swap, that hasn't made any observable difference in the dmesg output.

Out of curiosity, i have tried a random USB3 to USB-C adapter i had laying around and plugged that into the dock, this did end up producing a slightly different dmesg output, though i'm not sure if it's meaningful. I suspect the adapter must have its own pull resistors inside itself?

<code>
[ 1022.399148] usb 3-1: new full-speed USB device number 12 using xhci_hcd
[ 1022.522518] usb 3-1: device descriptor read/64, error -71
[ 1022.752558] usb 3-1: device descriptor read/64, error -71
[ 1022.982560] usb 3-1: new full-speed USB device number 13 using xhci_hcd
[ 1023.105529] usb 3-1: device descriptor read/64, error -71
[ 1023.335858] usb 3-1: device descriptor read/64, error -71
[ 1023.442720] usb usb3-port1: attempt power cycle
[ 1023.845854] usb 3-1: new full-speed USB device number 14 using xhci_hcd
[ 1023.846051] usb 3-1: Device not responding to setup address.
[ 1024.052601] usb 3-1: Device not responding to setup address.
[ 1024.259184] usb 3-1: device not accepting address 14, error -71
[ 1024.259264] usb 3-1: WARN: invalid context state for evaluate context command.
[ 1024.589199] usb 3-1: new full-speed USB device number 15 using xhci_hcd
[ 1024.589656] usb 3-1: Device not responding to setup address.
[ 1024.796009] usb 3-1: Device not responding to setup address.
[ 1025.002517] usb 3-1: device not accepting address 15, error -71
[ 1025.002609] usb 3-1: WARN: invalid context state for evaluate context command.
[ 1025.002681] usb usb3-port1: unable to enumerate USB device
</code>

sigprof, 2024-05-30:
No idea what's inside the adapter you tried, but that does not look good.  Does any USB2 device work in those ports?  (USB3 devices use a completely separate set of contacts and wires for data transfer, so it is possible that USB3 works when USB2 is broken, or vice versa.)

n-kremeris, 2024-05-31:
Yes, i have a USB2 kensington trackball attached to the dock, and previously had the model m attached to the dock using the original controller + cable via a ps2 to USB adapter. I also just tried an old usb2 flash drive which is also recognized OK.