qmk/qmk_firmware #23834: [Bug] Raspberry Pi Pico does not work until manually puting RUN pin to ground.

mnesarco, 2024-05-30:
### Describe the Bug

I am making a NON-SPLIT keyboard, everything works if after connecting the USB to PC, i momentarily connect pin RUN to GND (Restart the board). But if i just connect the USB cabe, nothing happens, the keyboard does not work. 

I have read a lot of issues, documentation and not clue, so i ask here.

### Keyboard Used

custom

### Operating System

Linux Mint 21.3 x86_64

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.1
Ψ QMK home: /xxx/kbd2/qmk/firmware
Ψ Detected Linux (Linux Mint 21.3).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.25.0
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2024-05-30:
What kind of controller board are you using, or, if you have the RP2040 chip on a custom PCB, what is the model of the crystal?  Apparently some crystals may need an increased startup delay with RP2040.  So you may try adding the following option to `config.h`:
```c
#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
```
(or maybe even 128 — there is one board like that in the repo)

mnesarco, 2024-05-31:
@sigprof i am using the original raspberry pi pico board. I will try your suggestion.

mnesarco, 2024-05-31:
I tried both:
```
#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 64
```
and
```
#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 128
```
none of them worked.

mnesarco, 2024-05-31:
This is my current config:

config.h
```c
#pragma once

#define UNICODE_SELECTED_MODES UNICODE_MODE_LINUX
#define TAPPING_TERM 250

#define MATRIX_ROWS 8
#define MATRIX_COLS 6

#define DEBUG_MATRIX_SCAN_RATE

#define UART_DRIVER SIOD1
#define UART_TX_PIN GP8
#define UART_RX_PIN GP9

#define PICO_XOSC_STARTUP_DELAY_MULTIPLIER 128
```

halconf.h

```c
#pragma once
#include_next <halconf.h>
#undef HAL_USE_SIO
#define HAL_USE_SIO TRUE
```

mcuconf.h
```c
#pragma once
#include_next <mcuconf.h>

#undef RP_SIO_USE_UART1
#define RP_SIO_USE_UART1 TRUE
```

rules.mk
```
ENCODER_MAP_ENABLE = no
ENCODER_ENABLE = no
CAPS_WORD_ENABLE = yes
UNICODE_ENABLE = yes
BOOTLOADER = rp2040
BOARD = GENERIC_RP_RP2040

UART_DRIVER_REQUIRED = yes
SPLIT_KEYBOARD = no

SRC += matrix.c
CUSTOM_MATRIX = lite

CONSOLE_ENABLE = yes
DEBUG_ENABLE = yes

```

sigprof, 2024-05-31:
> But if i just connect the USB cabe, nothing happens, the keyboard does not work.

Is the keyboard not detected as an USB device at all, or does it get detected normally but does not report any keypresses?  Check the `lsusb` output, and maybe the `dmesg` output for possible USB errors.

Also you can try flashing the `handwired/onekey/rp2040` firmware for testing purposes (it probably won't report any keypresses, because it tries to use only GP4 and GP5 for the switch “matrix”, but it should be detected as an USB keyboard).  If the board with that firmware does not get reliably detected as a keyboard, you probably need to look for some hardware issue; otherwise the problem is probably somewhere in your custom firmware code (or maybe the custom hardware — what is connected to the UART?).

sigprof, 2024-05-31:
> ```c
> bool data_available(void) {
>     uint32_t timeout = 0;
>     while (!uart_available()) {
>         timeout++;
>         if (timeout > UART_MATRIX_RESPONSE_TIMEOUT) {
>             uart_available();
>         }
>     }
>     return true;
> }
> ```

This code looks wrong — the inner `uart_available();` should probably be `return false;`, otherwise in the timeout case `matrix_scan_custom()` will still call `uart_read()`, which blocks indefinitely until some data arrives.  Maybe this happens immediately after the power-on reset, when your ESP32 board is not yet ready for some time.

And measuring the timeout in the number of loops is not very good, because the resulting timeout will vary greatly between different MCUs (and in the RP2040 case the performance might not be constant due to the XIP cache behavior).  You may use a timeout based on `timer_read()`, but in that case the minimum timeout that you can have is 1…2 ms; but this may be good enough if the timeout is just to avoid a complete lockup in case of some malfunction.

mnesarco, 2024-05-31:
> This code looks wrong — the inner `uart_available();` should probably be `return false;`, otherwise in the timeout case `matrix_scan_custom()` will still call `uart_read()`, which blocks [...]
OMG i have made so many changes and at some point removed the return by mistake. The correct code is:

```c
bool data_available(void) {
    uint32_t timeout = 0;
    while (!uart_available()) {
        timeout++;
        if (timeout > UART_MATRIX_RESPONSE_TIMEOUT) {
            return uart_available();
        }
    }
    return true;
}
```

The cycle based timeout was copied from another keyboard in the repo. And it works normally after restart. But as the function `matrix_init_custom` is never called, i think the problem is not in data reading, it must be something before even getting here. It never gets there.

sigprof, 2024-05-31:
Are you sure that `matrix_init_custom()` is never called?  You might not see the debug message because it lacks a final `\n`, or because some subsequent code locks up before the message could be sent over USB.

mnesarco, 2024-05-31:
How can i check that it is called? i added the \n at the end but nothing is printed in the console. Is it possible to add some delay before and after uart_init?

mnesarco, 2024-05-31:
I confirmed that something is blocking in `matrix_init_custom` or `matrix_scan_custom` because commenting the uart code it starts. So I have to figure out what is going on with the uart_init, maybe i need to add some delay before and after.

mnesarco, 2024-05-31:
@sigprof Thank you for your help. I have identified the culprit in `matrix_scan_custom`. Serial protocol is hard, it needs a lot of synchronization and timing. I have solved the problem playing with some checks and timeouts.

closed 2024-05-31 by mnesarco