qmk/qmk_firmware #23849: [Bug] wrong definition of digitizer endpoint

jhorology, 2024-06-02:
### Describe the Bug

https://github.com/qmk/qmk_firmware/blob/fa6d23235bf429446250cd5212e209d5fbfdbac2/tmk_core/protocol/chibios/usb_endpoints.c#L63-L65

I think this is a mistake. It should be `USB_ENDPOINT_IN_DIGITIZER` instead of `USB_ENDPOINT_IN_JOYSTICK`, shouldn't it? 

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

jhorology, 2024-06-02:
And also 5th parameter of  `QMK_USB_ENDPOINT_IN()` is missing.

KarlK90, 2024-06-02:
@jhorology thanks for reporting this error, I have opened #23854 to fix this.

closed 2024-06-03 by KarlK90