qmk/qmk_firmware #23850: [Bug] rgb matgrix effect bug.

itarze, 2024-06-02:
### Describe the Bug

https://github.com/qmk/qmk_firmware/blob/master/quantum/rgb_matrix/animations/pixel_fractal_anim.h  

This RGB effect does not limit the NO_LED and will cause access to the wild pointer, which will cause the MCU to malfunction.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2024-06-08:
I'm curious which controller this is for.  I can't replicate the MCU crash using an STM32F303 (using a moonlander, which does have a bunch of NO_LED values)

jiaxin96, 2024-07-06:
See https://github.com/qmk/qmk_firmware/pull/23348

jiaxin96, 2024-07-06:
same as https://github.com/qmk/qmk_firmware/pull/23348

itarze, 2024-07-23:
> I'm curious which controller this is for. I can't replicate the MCU crash using an STM32F303 (using a moonlander, which does have a bunch of NO_LED values)

Set NO LED to the left half of the keyboard RGB MATRIX， Like this.   
[image]

itarze, 2024-07-23:
And， This will slove it.   
```c
void rgb_matrix_set_color(int index, uint8_t red, uint8_t green, uint8_t blue) {
    if (index >= RGB_MATRIX_LED_COUNT) {
        return;
    }
    rgb_matrix_driver.set_color(index, red, green, blue);
}
```

itarze, 2024-07-23:
This problem was generated randomly because this lamp effect used random8(), and it may be that you were not able to reproduce it for this reason.

PR #23881 (2024-07-23): Check LED flags for Pixel Fractal effect

## Description

Refactor effect to account for LED flags

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #23850

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://doc [... cut]
closed 2024-07-23 by zvecr