qmk/qmk_firmware #23871: stdint.h: No such file or directory when compiling

luizkowalski, 2024-06-05:
### Issue Description

hey

I have GMMK2, whose config is this:

```c
#include QMK_KEYBOARD_H

// Decrease debounce because this keyboard is weird
#undef DEBOUNCE
#define DEBOUNCE 2

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
  /* Keymap _BL: Base Layer (Default Layer)
   */
[0] = LAYOUT(
  KC_ESC,   KC_1,     KC_2,     KC_3,     KC_4,     KC_5,     KC_6,     KC_7,     KC_8,     KC_9,     KC_0,     KC_MINS,  KC_EQL,   KC_BSPC,  KC_DEL,
  KC_TAB,   KC_Q,     KC_W,     KC_E,     KC_R,     KC_T,     KC_Y,     KC_U,     KC_I,     KC_O,     KC_P,     KC_LBRC,  KC_RBRC,  KC_ENT,   KC_PGUP,
  KC_CAPS,  KC_A,     KC_S,     KC_D,     KC_F,     KC_G,     KC_H,     KC_J,     KC_K,     KC_L,     KC_SCLN,  KC_QUOT,  KC_NUHS,            KC_PGDN,
  KC_LSFT,  KC_NUBS,  KC_Z,     KC_X,     KC_C,     KC_V,     KC_B,     KC_N,     KC_M,     KC_COMM,  KC_DOT,   KC_SLSH,  KC_RSFT,  KC_UP,    KC_END,
  KC_LCTL,  KC_LALT,  KC_LGUI,                                KC_SPC,                                 KC_RGUI,  MO(1),    KC_LEFT,  KC_DOWN,  KC_RGHT),

  /* Keymap _FL: Function Layer
   */
[1] = LAYOUT(
  KC_GRV,   KC_BRID,  KC_BRIU,    KC_F3,    KC_F4,    KC_F5,    KC_F6,    KC_F7,    KC_F8,    KC_F9,  KC_MUTE,  KC_VOLD,  KC_VOLU,   _______,  KC_INS,
  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,   _______,  _______,
  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,  _______,             _______,
  _______,  _______,  RGB_HUI,  RGB_HUD,  RGB_SPD,  RGB_SPI,  _______,  _______,  _______,  KC_MPRV,  KC_MPLY,  KC_MNXT,  _______,   RGB_VAI,  KC_HOME,
  _______,  _______,  _______,                                QK_BOOT,                                _______,  _______,  RGB_RMOD,  RGB_VAD,  RGB_MOD),
};

#define CAPS_POSITION 30

bool rgb_matrix_indicators_advanced_user(uint8_t led_min, uint8_t led_max) {
    if (IS_HOST_LED_ON(USB_LED_CAPS_LOCK)) {
        rgb_matrix_set_color(CAPS_POSITION, HSV_ORANGE);
    }
    return true;
}

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    // send CMD + NUBS on CMD + ESC
    if (keycode == KC_ESC && get_mods() & MOD_BIT(KC_LGUI)) {
        if (record->event.pressed) {
            register_code(KC_NUBS);
        } else {
            unregister_code(KC_NUBS);
        }
        return true;
    }
    return true;
}
```

When I try to compile this, I get this error:

```
❯ make gmmk/gmmk2/p65/iso:kowalski:flash
QMK Firmware 0.19.10
Makefile:401: keyboards/gmmk/gmmk2/p65/iso/rules.mk: No such file or directory
Making gmmk/gmmk2/p65/iso with keymap kowalski and target flash

arm-none-eabi-gcc (GCC) 14.1.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/gmmk/gmmk2/p65/iso/iso.c                                                      In file included from ./lib/chibios/os/hal/osal/rt-nil/osal.h:29,
                 from ./lib/chibios/os/hal/include/hal.h:30,
                 from ./platforms/chibios/drivers/wear_leveling/wear_leveling_efl_config.h:6,
                 from <command-line>:
/opt/homebrew/Cellar/arm-none-eabi-gcc/14.1.0/lib/gcc/arm-none-eabi/14.1.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
 [ERRORS]
```

I do believe I have the deps I need installed:

```
arm-none-eabi-gcc arm-none-eabi-binutils gmp isl libmpc lz4 mpfr xz zstd
osx-cross/arm/arm-none-eabi-gcc@9 gmp isl libmpc mpfr osx-cross/arm/arm-none-eabi-binutils
arm-none-eabi-gdb ca-certificates gmp mpdecimal mpfr openssl@3 python@3.12 readline sqlite xz
```

I'm on MacOS 14.5 and QMK Firmware 0.19.10

fauxpark, 2024-06-05:
You have the wrong ARM toolchain. The one in homebrew-core is broken and should not have been added...

luizkowalski, 2024-06-06:
will try that and report back, thanks!

melcz, 2024-06-06:
> following the QMK docs this time.

Could you clarify what versions and dependencies these are? The main setup instructions just refer to `qmk setup` which should've taken care of all the deps?

fauxpark, 2024-06-06:
@luizkowalski something in your environment is causing clang to be used instead of arm-none-eabi-gcc. It might be a problem with your brew install.

luizkowalski, 2024-06-07:
> Could you clarify what versions and dependencies these are? The main setup instructions just refer to `qmk setup` which should've taken care of all the deps?

```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/luiz/firmwares/qmk_firmware
Ψ Detected macOS 14.5 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.19.10
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

@fauxpark, that makes sense. Do you have any idea how I could debug this?

luizkowalski, 2024-06-07:
Ok, so I went to https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads and downloaded the toolchain, uninstalled the ones that `qmk setup` installed, added the bin folder I extracted from the toolchain to the path, fixed a couple of errors in the mapping (`LAYOUT_65_iso_blocker` and `host_keyboard_led_state`) and now it is working

indeed the deps installed by QMK from brew are broken

fauxpark, 2024-06-08:
> indeed the deps installed by QMK from brew are broken

No, this is an issue with your environment. Please do not use the toolchain directly from ARM.

The ARM and AVR toolchains that the CLI formula installs are from osx-cross, and are confirmed working. There is something wrong with your Brew install which you need to figure out and fix.

Your repo is also very old; the current version is 0.25.x, whereas you are on 0.19.x.

luizkowalski, 2024-06-08:
I nuked my homebrew and reinstalled everything

```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/luiz/qmk_firmware
Ψ Detected macOS 14.5 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.25.7
Ψ - Latest master: 2024-06-07 14:25:20 +0100 (e7a08ef1a9) -- Fix broken link in PR checklist (#23877)
Ψ - Latest upstream/master: 2024-06-07 14:25:20 +0100 (e7a08ef1a9) -- Fix broken link in PR checklist (#23877)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-06-07 14:25:20 +0100 (e7a08ef1a9) -- Fix broken link in PR checklist (#23877)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

everything is working now without the toolchain from ARM. 

thanks a lot, folks!

closed 2024-06-08 by luizkowalski