qmk/qmk_firmware #23895: [Feature Request] Speedup setup

thewh1teagle, 2024-06-11:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

When using `qmk setup` currently, it downloads numerous repositories and libraries, many of which may not be necessary for my usage. For example, I primarily use the Pico microcontroller, but it seems to clone libraries for other microcontrollers as well. Consequently, `qmk setup` takes a considerable amount of time.

Additionally, I'm interested to know if it's feasible to streamline the codebase of a custom `qmk` keyboard. I'd like to retain only the custom layout in the layouts folder and the custom keyboard in the keyboards folder. Are all the additional folders such as drivers, lib, platforms, utils, users, and others necessary in every `qmk` git repo, or can they be somehow streamlined to utilize defaults from qmk?

Thanks in advance!

tzarc, 2024-06-11:
Trying to do so is like steering a fully-laden container ship. Doing so requires a lot of effort, and in the case of QMK, likely several years of concerted effort chipping away at such an endeavour.

If you'd prefer to just keep your own keymap and/or layout, we do support an [external userspace](https://docs.qmk.fm/newbs_external_userspace) which allows you to keep your own repository with just your keymaps, and let GitHub do the build for you.

Also, It's currently not yet documented (there's a billion other things to do!), but you can use GitHub codespaces to set up the userspace build environment on their servers, and perform build there.

At this stage though -- your initial request to break things out piecemeal -- is unlikely to occur I'm afraid. It's much lower on the priority list than most.

thewh1teagle, 2024-06-11:
Thanks for the insight into the challenges of managing qmk. I understand it's a bit like steering a fully-laden container ship – no easy task.

```console
qmk_huge_repo/keyboards/custom/
├── config.h
├── info.json
├── keymaps
│   └── default
│       ├── keymap.c
│       └── keymap.h
└── rules.mk
```

As you can see, I actually use only 5 files for this custom keyboard, including custom hooks, and I probably use the default layout. Instead of focusing on them, I have a very big repo, and it's very hard to track git changes and update qmk core. Do you have a suggestion or even a workaround for now? (external userspace supports custom keyboard like the files mentioned?) Maybe the qmk cli can improve and retrieve defaults in case of a clean repository.

The best scenario I can think of is qmk cli detecting the config file and adding most of the things inside a dot folder so I can really focus on what's important. Additionally, a qmk subcommand such as qmk update to update all qmk core components would be helpful.

tzarc, 2024-06-11:
Then unfortunately userspace isn't really an option for you -- keyboards aren't officially supported as the complexities dealing with paths within the repo exponentially increase the search space.

You're better off making a branch and periodically merging QMK's master into your working branch. Nothing you're modifying affects core code, so keeping up to date with merges will be trivial.

closed 2024-09-09 by tzarc