qmk/qmk_firmware #23900: [Bug]

OmegaLambda1998, 2024-06-12:
### Describe the Bug

For certain rgb matrix effects, when all (or most) LED's are blank, random keys will flash a random colour. This seems to (subjectively) be focused on keys in the bottom right quadrant (last 2-3 rows, and coloums to the right of space) but I've seen this behaviour across the entire keyboard. This behaviour seems similar to #23619 though with different triggers, as I see the flashing when no effect occurs, rather than when an effect reaches a particular part of the matrix.

The affected rbg matrix effects include:

- breathing (in between "breaths")
- band_val (but not band_pinwheel_val or band_spiral_val)
- typing_heatmap, but only if I haven't typed in a while, i.e when the "heatmap" dies down. I need to type a few words to get enough "activity" to stop the bug. Typing that sentence for instance was enough.
- digital rain, for the first ~5 seconds
- solid_reactive_[simple, wide, multiwide, cross, multicross, nexus, multinexus, splash, multispash, solid_splash, solid_multisplash] all have the bug in between typing, when all keys are blank.O

### Keyboard Used

annepro2/c18

### Operating System

Arch Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/*redacted*/User/Development/Bash/annepro2/qmk_firmware
Ψ Detected Linux (Arch Linux).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ - Latest master: 2024-06-12 10:24:04 +1000 (eb30f377d8) -- Simplify keymap to isolate bug
Ψ - Latest upstream/master: 2024-06-10 01:23:25 +0100 (8b5cdfabf5) -- Re-implement `eeprom_write_qword` as define (#23890)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-06-07 14:25:20 +0100 (e7a08ef1a9) -- Fix broken link in PR checklist (#23877)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

AnnePro-Shine (https://github.com/OpenAnnePro/AnnePro2-Shine)

### Additional Context

I have an Anne Pro 2 (C18) flashed with almost the default keymap (see https://github.com/OmegaLambda1998/qmk_firmware/tree/master/keyboards/annepro2/keymaps/OmegaLambda). I then flash AnnePro-Shine  to get lighting functionality. I have flagged this issue with the OpenAnnePro team as well.

bombonatti, 2024-10-09:
Same issue here, a workaround that I found is to set stick led on space bar (did not look why it removes glitchy).

Basically, I declare a layer (I named as MASKSET) and set a value for space bar:
```
 [MASKSET] = LAYOUT_60_ansi( /* MASKSET */
    _______, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______, _______,
    _______, _______,    _______,    _______,    _______,    _______, _______, _______, _______, _______, _______, _______, _______,  _______,
    _______, _______,    _______,    _______,    _______,    _______, _______, _______, _______, _______, _______, _______, _______,
    _______,             _______,    _______,    _______,    _______, _______, _______, _______, _______, _______,  _______,  _______,
    _______, _______,    _______,                                     KC_GRV,                   _______, _______, _______, _______
 ),
```

on function layer_state_set_user, I added a loop to stick led color:
```
    for(i=0; i<MATRIX_ROWS; i++) {
        for(j=0; j<MATRIX_COLS; j++) {
            if(keymaps[MASKSET][i][j]!=_______)
                ap2_led_sticky_set_key(i,j, color[0]);
        }
    }
```
I am attaching my keymap.c for reference, maybe it may help someone to debug/fix.

[keymap.zip](https://github.com/user-attachments/files/17318968/keymap.zip)

SgoSkz, 2025-01-17:
https://github.com/OpenAnnePro/AnnePro2-Shine/blob/master/main.c#L29
https://github.com/qmk/qmk_firmware/blob/master/keyboards/annepro2/annepro2.c#L32

For anyone still stuck on this error, changing these two lines from 115200 to 57600 seems to fix this issue for me. 

keymap: default
board: C18
No shine or qmk modifications aside from those listed above

realfrostbit, 2025-01-29:
Having the same bug, I can confirm that @bombonatti workaround did indeed help, but the solution to halve both UART baud rates provided by @SgoSkz solves this problem reliably for me. Many Thanks!

Fede-26, 2026-11-11:
The 57600 workaround works, but the baud rate is not the cause, and there is a fix that keeps 115200.

The problem is on the **LED MCU**, not on the QMK side. `mainCallback` in Shine's `source/matrix.c` runs off an 80 kHz timer at interrupt priority **1**, while that chip's UART is at priority **2**. On Cortex-M0+ lower is higher, so **the UART cannot preempt the PWM interrupt** — every microsecond spent there is a microsecond the receive register is not being read.

And that interrupt was calling the profile callbacks directly: all 70 keys, with `hsv2rgb` per key on the reactive profiles. Halving the baud rate doubles the time a character takes to arrive, which is why it papers over the problem.

Fix in OpenAnnePro/AnnePro2-Shine#66: the profile callbacks move to a thread below the UART reader, the per-channel colour correction is precomputed instead of being recomputed ~15k times a second, and the two divisions left in the interrupt are gone (this core has no divide instruction, so each was a call into `__aeabi_uidiv`). `mainCallback` now makes no function calls at all.

**Verified on C18 at 115200**, against a QMK build with the stock baud rate: no stray lit keys on a near-dark matrix, which is the symptom described here. This needs a reflash of the LED MCU, not of QMK.

Tracked on the Shine side as OpenAnnePro/AnnePro2-Shine#61.

---

Separately, and unrelated to the LEDs: while chasing this I found that this keyboard's matrix scans at exactly 200 Hz, because `wait_us()` falls back to a 1 ms tick sleep. That one is #26573, and it is the likelier explanation for the chattering reports on this board — worth keeping the two apart, they look similar but are different bugs.