qmk/qmk_firmware #23907: [keyboard] Add Framework 16

PandorasFox, 2024-06-13:
### Issue Description

The modular [framework laptops](https://frame.work/)'s 16-inch model has a qmk-compatible keyboard in it, but the firmware doesn't appear to have been upstreamed.

The [firmware repo](https://github.com/FrameworkComputer/qmk_firmware/tree/v0.2.9) - latest tagged release (6mo ago); there has not been any further development since then.

The keyboard(s) here are a bit weird (typically); there's both the main board and the keypad/macropad, which I believe are compiled and flashed separately. Upstreamed support here would be really beneficial :)

tzarc, 2024-06-13:
This was attempted by the manufacturer already: https://github.com/qmk/qmk_firmware/pull/21722

Unfortunately it was left to go stale by inaction -- you might be best off asking the manufacturer to resume their attempts at upstreaming.

zvecr, 2024-06-13:
From the issue templates,

> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
Your best bet is to take the initiative, add support, then submit a PR yourself.

closed 2024-06-13 by zvecr

PandorasFox, 2024-06-13:
> From the issue templates,
> 
> > Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
> > Your best bet is to take the initiative, add support, then submit a PR yourself.

https://docs.qmk.fm/configurator_step_by_step

the docs explicitly say to file an issue if there's not an active pull request. The pull request is specifically stale. Would it not be best to leave this issue open so it's actually searchable until this is actually resolved, even if it's somewhat blocked?

zvecr, 2024-06-13:
> the docs explicitly say to file an issue if there's not an active pull request

That line would need to be replaced with something that reflects the current process.

> Would it not be best to leave this issue open so it's actually searchable until this is actually resolved

We actively close these issues now as it bloats the issue backlog with tickets that can not be actioned.