qmk/qmk_firmware #23909: Any way to customise the RGB Ligting (not a matrix)?

ipg0, 2024-06-13:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Currently there is a well-documented and relatively straightforward way to implement custom RGB matrix effects/animations, but no complementary functionality for non-matrix RGB lighting (a single LED strip).

I have not been able to find any info on creating custom non-matrix RGB lighting modes or hooking into/overriding any functionality regarding non-matrix RGB lighting, in a similar way to how matrix effects are customised. If there is a way of doing so that is considered "proper", the documentation should be updated to include it, otherwise, I think, this functionality should be implemented and documented afterwards.

Anyway, I would appreciate if anyone let me know

ipg0, 2024-06-21:
I'm in the process of implementing this functionality right now, will open a PR once I'm done

zvecr, 2024-07-31:
Given the current state of rgblight, and its maintenance issues, this is unlikely the direction we would want to go in.

The suggestion would be to just use RGB Matrix instead, as this supports the additional functionality you are after.

closed 2024-11-29 by github-actions[bot]