qmk/qmk_firmware #23934: [Bug] Media keys' functions (⏯️, ⏮️, ⏭️, 🔉, 🔊, 🔇) don't get flashed to my keyboards in newly-generated .hex files

adiabatic, 2024-06-16:
### Describe the Bug

I've been using media keys on my Ergodoxen EZ for a while. Like on the default layout, my layout, [Zweihänder](https://github.com/adiabatic/zweihander/blob/main/keyboards/ergodox_ez/base/keymaps/zweihander_macos/keymap.c), uses the following keys:

- KC_MPLY
- KC_MPRV
- KC_MNXT
- KC_VOLD
- KC_VOLU
- KC_MUTE

However, after flashing my most recent update to my two newest keyboards, volume-up and volume-down didn't work. On either my Mac or my iPad.

By "don't work" I mean "they don't do anything when I press them". They're not acting like `KC_TRNS`.

## Trying to get it to work

I rebooted my Mac. The volume keys still didn't work.

I tried flashing a slightly older .hex file from a couple days ago — the one that added two caps lock keys. The media keys didn't work even with this older .hex file.

## Not-recently-flashed keyboards work OK

I also pulled out my old Ergodox EZ that hadn't seen the light of day in the past couple weeks (it's been in a cabinet). I plugged it into my iPad and the media keys all worked fine.

## A two-month-old .hex file works

I flashed a two-month-old .hex file to my keyboard and the media keys are back to working again. This was the change in the file between the two-month-old hex file and the next one:

https://github.com/adiabatic/zweihander/commit/8e5d17b332e57400a5d1396463ea7dceb760a333

tl;dr add two capslocks; comment out a `custom_keycodes` and a `process_record_user` function

## Newly-made Ergodox EZ default layout doesn't work

I set up `qmk` in a `venv` like my Debian machine told me to do, deviating from the instructions on the QMK website slightly. I was able to generate a default .hex layout and flash it by running `../bin/qmk flash --keyboard ergodox_ez --keymap default`. The media keys didn't work on that keyboard after doing this.

## Speculations on what the problem could be

I had a look at <https://github.com/qmk/qmk_firmware/commits/master/> and nothing between now (Sunday, June 16, 2024 at 3:35:48 AM `America/Los_Angeles`) and June 12 or so looks like it would have mucked up media keys on my keyboard. Then again, I could be very wrong about this.

## Files I tried flashing

[ergodox_ez_base_zweihander_macos-2024-04-14.hex.zip](https://github.com/user-attachments/files/15857486/ergodox_ez_base_zweihander_macos-2024-04-14.hex.zip) (media keys still work here)

[ergodox_ez_base_zweihander_macos-2024-06-14-two-more-capslocks.hex.zip](https://github.com/user-attachments/files/15857498/ergodox_ez_base_zweihander_macos-2024-06-14-two-more-capslocks.hex.zip) (first/oldest "media keys stop working" .hex)

## …

Any ideas what could be going wrong?

### Keyboard Used

ergodox_ez/base

### Operating System

macOS 14.5, iPadOS 17.5.1, Debian Stable (12?) running GNOME 3

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/comatoast/Projects/Others/qmk_firmware
Ψ Detected macOS 14.5 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.25.8
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.1.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

fauxpark, 2024-06-16:
Extrakey is probably not enabled, I can see the other EZ revisions (glow & shine) have it enabled under `features.extrakey` in their respective `keyboard.json`s but not `base`.

adiabatic, 2024-06-16:
I need to add a JSON file alongside my `keymap.c` now…?

fauxpark, 2024-06-16:
No.

It looks like the `rules.mk` -> `info.json` migrations for `base` were missed in this PR: https://github.com/qmk/qmk_firmware/pull/23512

PR #23935 (2024-06-16): `ergodox_ez/base`: Add missing `features` object

## Description

Fixes #23934.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* 

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist) and ha [... cut]
closed 2024-06-16 by fauxpark