qmk/qmk_firmware #23946: Language related send_string issue

zadorid88, 2024-06-17:
### Issue Description

Hi,

First time ever trying to code something, and I am stuck. 

I need a macro which does two simple key taps: "LALT" first, than "ő".
I used 
#include "keymap_hungarian.h"
#include "sendstring_hungarian.h"
but the string doesn't seem to know this "ő" letter, no matter what I do.

I can program the letter to a dedicated key using "HU_ODAC", which is the code from the "keymap_hungarian.h", but it doesn't work in sendstring.

I tried:
SEND_STRING(SS_TAP(X_LALT)"ODAC")
SEND_STRING(SS_TAP(X_LALT)"Ő")
SEND_STRING(SS_TAP(X_LALT)"HU_ODAC")
SEND_STRING(SS_TAP(X_LALT)"X_ODAC")
but no luck.

Please help!

fauxpark, 2024-06-17:
Please read the docs: https://docs.qmk.fm/features/send_string#send-string
Only ASCII characters are supported.

fauxpark, 2024-06-17:
To clarify, the sendstring header is mainly to map the ASCII character set to the appropriate keycodes for the specific language. It does not allow you to use any character from that language in sendstrings.

You can still call `tap_code()` (and `tap_code16()` for modified keycodes) with the keycodes from the keymap header. For short sequences like Alt+ő this is preferable to sendstring anyway as that's mostly for typing out longer things like sentences.

closed 2024-06-17 by fauxpark

zadorid88, 2024-06-17:
Wow, this worked, thank you very much! :)