qmk/qmk_firmware #23968: [Bug] compile_commands.json breaks VSCode clangd intellisense by defining non-existent macros

BrknRules, 2024-06-20:
### Describe the Bug

`compile_commands.json` includes the following macro definitions:
`-D __has_include(STR)=__has_include__(STR)` and `-D __has_include_next(STR)=__has_include_next__(STR)`

This causes clangd to break VSCode's intellisense, emitting `macro not defined` errors.

Removing these flags from the `compile_commands.json` ended up fixing the issue for me.

### Keyboard Used

keebio/quefrency/rev4

### Link to product page (if applicable)

https://keeb.io/products/quefrency-rev-4-65-split-staggered-keyboard?_pos=13&_sid=e65f328e4&_ss=r

### Operating System

MacBook Pro M1 Pro macOS Sonoma 14.5

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

QMK Toolbox

### Additional Context

`qmk compile --compiledb -kb keebio/quefrency/rev4 -km default` was run first.
`clangd` version = 18.1.3 (VSCode clangd extension latest install)

willhack, 2024-07-01:
Same issue here. Compiling and flashing run flawlessly, there seems to be something odd with the linter/LSP.

For sanity I tried a fresh install of VSCode as well, making sure to blow out any existing settings and just following the [QMK guides](https://docs.qmk.fm/other_vscode) to the letter.

### Keyboard Used
draculad - but I've notice the bug on whatever keyboard I open.

### Operating System
MacBook Air M2 macOS Sonoma 14.5

### qmk doctor Output
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/will/repos/qmk_firmware
Ψ Detected macOS 14.5 (Intel).
Ψ Userspace enabled: False
Ψ Git branch: willhack
Ψ Repo version: 0.24.7
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest willhack: 2024-06-28 00:01:49 -0400 (a2dac72f4a) -- auto-migrate
Ψ - Latest upstream/master: 2024-06-27 13:37:41 -0400 (9d02ac37f7) -- Add Nyquist Rev. 5 (#23971)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-06-27 13:37:41 -0400 (9d02ac37f7) -- Add Nyquist Rev. 5 (#23971)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
 ....
Ψ QMK is ready to go, but minor problems were found
```

### Other keyboard-related software installed
`None`

### Additional Context
Also ran `qmk generate-compilation-database` resulting in: 
```
Ψ Gathering build instructions from gmake -r -R -f builddefs/build_keyboard.mk -s -n KEYBOARD=draculad KEYMAP=gotham KEYBOARD_FILESAFE=draculad TARGET=draculad_gotham INTERMEDIATE_OUTPUT=.build/obj_draculad_gotham VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"
```

Screenshot showing various error messages resulting.
[image]

---

`qmk lint` goes off without a hitch
```Ψ Lint check passed!```

[compile_commands.json](https://github.com/user-attachments/files/16047061/compile_commands.json)

tzarc, 2024-07-01:
There's obviously something going on with the compiler here; if I do `qmk compile --compiledb -kb draculad -km default` I don't get anything with `-D __has_include(STR)=__has_include__(STR)` nor `-D __has_include_next(STR)=__has_include_next__(STR)`.

Note that clang is not a participant at the point where `compile_commands.json` is generated -- that's a combination of parsing the `make` output and inferring values from the defaults supplied by the compiler by preprocessing an empty file with the compiler arguments set up to build for the target architecture.

I think the best we can do is to blacklist such entries.

tzarc, 2024-07-01:
Can I just confirm that removing all instances of the `__has_include` and `__has_include_next` defines from `compile_commands.json` actually fixes things?

My local environment doesn't export them, and I correctly get the following:
[image]

tzarc, 2024-07-01:
Yeah looks like it's an old compiler issue -- if I reissue the same command within the `ghcr.io/qmk/qmk_cli:latest` container (with gcc 8.x) then I get the same output in `compile_commands.json` as well as the deficient loading within VScode.

willhack, 2024-07-01:
> Can I just confirm that removing all instances of the `__has_include` and `__has_include_next` defines from `compile_commands.json` actually fixes things?
> 

Not sure if this solves OPs issues, but I'm still seeing erroneous linting errors. I've manually removed all instances of `-D __has_include(STR)=__has_include__(STR)` and `-D __has_include_next(STR)=__has_include_next__(STR)`. Also switched to testing on the `default` draculad keymap just for a little more sanity.

[image]

Updated [compile_commands.json](https://github.com/user-attachments/files/16047294/compile_commands.json)

tzarc, 2024-07-01:
Did you restart the language server after regenerating the `compile_commands.json`?

willhack, 2024-07-01:
Yep! Even blew out the `.cache` and restarted VSCode. That being said in the `clangd` output I see the following:
```
E[22:26:40.434] IncludeCleaner: Failed to get an entry for resolved path : No such file or directory
...
I[22:26:40.449] Indexed c17 standard library (incomplete due to errors): 2506 symbols, 24 filtered
```
Could that `IncludeCleaner` have something to do with it? Not sure how to debug what resolved path it's looking for

tzarc, 2024-07-01:
Honestly not sure; I've been using gcc 14 for a while and don't strike any of these issues. I'll try and replicate on an older system later today.

tzarc, 2024-07-01:
If you've got some familiarity with setting up things on your `$PATH`, you could try the same toolchain and see if it helps:
https://github.com/qmk/qmk_toolchains/releases/tag/v14.1.0-3

willhack, 2024-07-01:
> If you've got some familiarity with setting up things on your `$PATH`, you could try the same toolchain and see if it helps: https://github.com/qmk/qmk_toolchains/releases/tag/v14.1.0-3

Not entirely sure how to switch the toolchain via the path.
Just did quite a bit of googling and switched from apple clang to the brew/gnu versions of gcc and g++ ---
```
❯ gcc --version
gcc-14 (Homebrew GCC 14.1.0_1) 14.1.0
```

but I'm still seeing the errors and `compile_commands.json` hasn't changed. 

Not entirely sure where to poke around at next, but happy to keep trying things!

tzarc, 2024-07-01:
> Not entirely sure how to switch the toolchain via the path. Just did quite a bit of googling and switched from apple clang to the brew/gnu versions of gcc and g++

I’d suggest rolling this back — you’ve changed the compiler targeting your MacBook — I was suggesting changing the compiler targeting the chip for QMK. Two completely different beasts. 

For now I’d hold off on making any more changes; I’m seeing if I can do something with toolchain management baked into the QMK CLI itself.

willhack, 2024-07-01:
Yikes, rolled back! thank you for the support on all of this, wild how much goes into some "simple" linting and syntax highlighting

BrknRules, 2024-07-02:
> > Can I just confirm that removing all instances of the `__has_include` and `__has_include_next` defines from `compile_commands.json` actually fixes things?
> 
> Not sure if this solves OPs issues, but I'm still seeing erroneous linting errors. I've manually removed all instances of `-D __has_include(STR)=__has_include__(STR)` and `-D __has_include_next(STR)=__has_include_next__(STR)`. Also switched to testing on the `default` draculad keymap just for a little more sanity.
> 
> <img alt="Screenshot 2024-06-30 at 10 17 41 PM" width="952" src="https://private-user-images.githubusercontent.com/18036720/344515304-48f93aa6-6c65-4bed-a90f-66727dd80577.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk5MTkzNTQsIm5iZiI6MTcxOTkxOTA1NCwicGF0aCI6Ii8xODAzNjcyMC8zNDQ1MTUzMDQtNDhmOTNhYTYtNmM2NS00YmVkLWE5MGYtNjY3MjdkZDgwNTc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAyVDExMTczNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIwMGRkMWU5OGY4YTBkNzgwMmFjYmI5ZjFlMWE4MzdlMTk2ZTU3NjM4ZDRiMTcxODhmNDE3YzJiZWMwMDQ0MWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rNnU6KqIZBkTWWFwcZcu1aQw3R1DcAOmmDOeNo05B_Q">
> Updated [compile_commands.json](https://github.com/user-attachments/files/16047294/compile_commands.json)

I only removed these specific flags because VSCode was throwing `macro not defined` errors after I drilled down and tried to find the type definitions in the codebase. If what @tzarc says is true, our problem may be a symptom of the compiler inferring differing values from the defaults based on some difference between our machines.

Maybe find which header files your types are defined in and checking if they're included in `compile_commands.json` correctly? As a temporary fix.

Jacfger, 2024-10-07:
My clangd is also acting up in vscode, but I'm not so sure how you're getting the macro not found problem. Right now what I saw is simply `Too many errors emitted, stopping nowclang(fatal_too_many_errors)`. LOL

But I do find the `has_include` in the compile_commands and removed them. Though it didn't do anything.

I'm also using MacOS, more specifically OSX 14.7

Abraxas-Knister, 2024-11-11:
I just learned that `qmk generate_compilation_database` exists and then stumbled across this error.

Herschenglime, 2024-12-18:
While I don't have the offending macro definitions from the original issue, I still run into the error where the compilation database generated by `qmk generate_compilation_database` doesn't work with clangd (in Emacs). I'm unsure if there's been any fixes in newer versions of QMK as I'm using the [ZSA Fork](https://github.com/zsa/qmk_firmware) for my Moonlander, but I found an alternative method that works for me!

The [Bear tool](https://github.com/rizsotto/Bear) generates a compile_commands.json by intercepting the compiler commands used by a command. After installing, you can run `bear -- qmk compile` from the base `qmk_firmware` directory and it should create a functional `compile_commands.json`. Now all of the LSP features work as expected and it doesn't show errors for keycodes, QMK functions, and so on.

I hope this helps!

wagerfield, 2025-04-04:
Same issue. I tried using `bear -- qmk compile` as per @Herschenglime's suggestion, but without success.

@tzarc manually removing `-D __has_include(STR)=__has_include__(STR)` and `-D __has_include_next(STR)=__has_include_next__(STR)` from `compile_commands.json` as per @willhack's [suggestion above](https://github.com/qmk/qmk_firmware/issues/23968#issuecomment-2198985538).

FWIW I followed the steps in the [VSCode guide](https://docs.qmk.fm/other_vscode#debugging-arm-mcus-with-visual-studio-code) to the letter. I'm on Mac.

PR #25180 (2025-04-21): Fix for `.clangd`.
## Description

Mainly discussed on #23968, includes suggested fix on that issue and was also confirmed separately be someone afflicted by the problem on Discord.

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #23968

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventi [... cut]
closed 2025-04-21 by tzarc