qmk/qmk_firmware #24029: Can I overwrite the host.c file of tmk?

KeyMagicHorse, 2024-07-01:
### Issue Description

 Why do I want to rewrite the host.c file inside keyboard? 
 Answer: Because I am developing a Bluetooth module with nkro functionality, but nkro functionality is not implemented in bluetooth.c and host.c. 
 qmk community response required: 
 If my bluetooth module supports nkro and via data forwarding (via bluetooth serial), how do I modify qvod's bluetooth driver to enable nkro and via data sending and receiving? 
 
  I said my train of thought. 
 For nkro: Just implement [void bluetooth_send_nkro(report_nkro_t *report);] in bluetooth.c. 
 For via: 
 I think via should not directly use the raw_hid.h this file interface, but need to interface out, in the host.c to do hid raw data read and write. 

What kind of trouble does qmk get into if I try to implement my ideas?

fauxpark, 2024-07-01:
It is not possible right now.
If you are looking to make a Bluetooth keyboard you should look at ZMK instead.

closed 2024-07-01 by fauxpark

KeyMagicHorse, 2024-07-01:
> It is not possible right now. If you are looking to make a Bluetooth keyboard you should look at ZMK instead.

No, I have implemented the Bluetooth module. Now I only need to modify host.c, bluetooth.c and via.c to implement it. And I made a via software that adapts to the bluetooth module.