qmk/qmk_firmware #24037: Links to coding conventions broken

thomasa88, 2024-07-03:
### Issue Description

Go to https://github.com/qmk/qmk_firmware/blob/869b7d9ae63f2d5990934f938e44ecfe6d1d9e1f/docs/contributing.md

Try to following the links to the coding conventions:

> 404 - page not found
The 869b7d9ae63f2d5990934f938e44ecfe6d1d9e1f branch of qmk_firmware does not contain the path docs/coding_conventions_c.

https://github.com/qmk/qmk_firmware/blob/869b7d9ae63f2d5990934f938e44ecfe6d1d9e1f/docs/coding_conventions_c
https://github.com/qmk/qmk_firmware/blob/869b7d9ae63f2d5990934f938e44ecfe6d1d9e1f/docs/coding_conventions_python

tzarc, 2024-07-03:
The docs are written to target https://docs.qmk.fm, not through GitHub.
https://docs.qmk.fm/contributing has valid, working links to both pages, for instance.

thomasa88, 2024-07-03:
Oops, my mistake.

closed 2024-07-03 by thomasa88