qmk/qmk_firmware #24073: Compile but layout UNCHANGED

masar3141, 2024-07-06:
### Issue Description

Hello, i use rp2040 pro micro, custom layout, had been working great so far but after some consecutives flash the layout doesnt update anymore. I tried qmk clean with no succes. Also i tried with succes to flash the board with a blink.uf2 to be sure it wasnt the fault of the board.

Compiling and flashing doesnt raise any error

tzarc, 2024-07-07:
Did you enable VIA?

https://docs.qmk.fm/faq_keymap#my-keymap-doesn-t-update-when-i-flash-it

masar3141, 2024-07-07:
That was totally the issue. Thank you for your help. I don't get how i didn't end up on this before. Maybe it should be in the debbugging/troubleshooting section? Anyway, thanks again.

closed 2024-07-07 by masar3141