qmk/qmk_firmware #24074: [Bug] Could not save userspace file: ['splitkb/aurora/sofle_v2/rev1', '0xErnie'] is not valid under any of the given schemas

0xErnie, 2024-07-07:
### Describe the Bug

Following [README.md in qmk_userspace](https://github.com/qmk/qmk_userspace/blob/0f10c50b4cf01d4dad4e9d19d1655207b37fb81e/README.md) creating of a new keymap in my userspace failed with an arbitrary error.

```
qmk new-keymap -kb splitkb/aurora/sofle_v2/rev1

Ψ Generating a new keymap

Name Your Keymap
Used for maintainer, copyright, etc

Your GitHub Username?  [Redacted] 0xErnie
Ψ Created a new keymap called 0xErnie in: /home/redacted/repos/qmk_userspace/keyboards/splitkb/aurora/sofle_v2/rev1/keymaps/0xErnie.
Ψ Compile a firmware with your new keymap by typing: qmk compile -kb splitkb/aurora/sofle_v2/rev1 -km 0xErnie.
☒ Could not save userspace file: ['splitkb/aurora/sofle_v2/rev1', '0xErnie'] is not valid under any of the given schemas

Failed validating 'oneOf' in schema['properties']['build_targets']['items']:
    {'oneOf': [{'$ref': '#/keyboard_keymap_tuple'},
               {'$ref': '#/json_file_path'}]}

On instance['build_targets'][0]:
    ['splitkb/aurora/sofle_v2/rev1', '0xErnie']
```

### Keyboard Used

splitkb/aurora/sofle_v2/rev1

### Operating System

WSL

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/kiwicola/qmk_firmware
Ψ Detected Linux (WSL, Ubuntu 22.04.3 LTS).
Ψ Testing userspace candidate: /home/kiwicola/repos/qmk_userspace -- Valid `qmk.json`
Ψ QMK userspace: /home/kiwicola/repos/qmk_userspace
Ψ Userspace enabled: True
Ψ Git branch: master
Ψ Repo version: 0.25.10
Ψ - Latest master: 2024-07-06 18:21:11 +1000 (4fbb395602) -- `ano`: fix layout name (#24067)
Ψ - Latest upstream/master: 2024-07-07 01:58:04 +0200 (67d512e639) -- Fix enter for two ANSI layouts on the TKD Cycle7 (#24070)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-07-06 18:21:11 +1000 (4fbb395602) -- `ano`: fix layout name (#24067)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-07-07:
Keymap names require lowercase characters.

0xErnie, 2024-07-07:
The prompt explicetly asks for my github username, which does contain upercase letters.

noroadsleft, 2024-07-07:
This is a relic of sorts from when we allowed user keymaps in the repository. We had a couple instances where a user would have e.g. `README.md` and `readme.md` in the same directory, which works fine on Linux but can completely break in Git clones on MacOS or Windows. We requested keymap names to match GitHub usernames because it made maintenance easier for us if we had to edit a user's keymap. (Reading a directory name is easier than checking through the Git history.)

We no longer allow user keymaps in the repository, but the checks from that era are still around in places.

0xErnie, 2024-07-07:
The prompt can be found at https://github.com/qmk/qmk_firmware/blob/67d512e639806b24e68cb35170789f614e779638/lib/python/qmk/cli/new/keyboard.py#L151 and https://github.com/qmk/qmk_firmware/blob/67d512e639806b24e68cb35170789f614e779638/lib/python/qmk/cli/new/keymap.py#L30

I would suggest would be better to just ask for a name without those newlines:

```
{fg_yellow}Name Your Keymap{style_reset_all} (lowercase only): """
    return question(prompt)
```

tzarc, 2024-07-08:
Just for clarity, we're having an internal discussions as to whether or not we relax the lowercase requirement in userspace repositories -- ultimately the owner of the userspace can make their own decisions. Will keep you posted when a decision is made.