qmk/qmk_firmware #24088: [Bug] Cirque will override mouse button status in relative mode even when CIRQUE_PINNACLE_TAP_ENABLE is not set.

kbjunky, 2024-07-11:
### Describe the Bug

I have mouse buttons under my thumb operated keys on my [keyboard](https://github.com/kbjunky/Chordie/blob/64c9026e4cbfe07542eb9456cd11550bd3cb5a71/src/chordie/keymaps/chording/keymap.c#L244). Then in [pointing_device_task_user ](https://github.com/kbjunky/Chordie/blob/64c9026e4cbfe07542eb9456cd11550bd3cb5a71/src/chordie/chordie.c#L92) mouse keys status is already changed to what Cirque is putting into it's `mouse_report` thus overriding my own report. Because of that I can't do click/drag because Cirque will set mouse button status to 0 thus making **mouse key up event**. I've pinned down the issue to [this line](https://github.com/qmk/qmk_firmware/blob/efa5b30cfdc0649c18ba5315a11cb3294c928dbd/quantum/pointing_device/pointing_device_drivers.c#L355). This only happens in **relative mode**. 

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2024-07-21:
Are you using `pointing_device_handle_buttons` at all? 

If not, you may be able to improve things by doing so.

kbjunky, 2024-07-22:
Hi Drashna, thank you for your response. That's not the issue I am having. I might have not present the problem correctly. What I am asking is, is it possible to distinguish between touching the Cirque for mouse move and taping it for mouse click and not changing the mouse button state when there was just the move motion. Now it will reset mouse button states if there's only touch detected, as in motion. Or not change mouse button states when there's no `CIRQUE_PINNACLE_TAP_ENABLE` set. The problem I am having is that when I send the custom mouse button state Cirque will override it. For example I'll send LMB down and then Cirque in it's report will send LMB up by zeroing all the button states but I might still be holding LMB. I have "fixed" it by storing the custom mouse button states and restoring them in `pointing_device_task_user()` but I think it would be better to just not change the state when there was no interaction related to mouse buttons.