qmk/qmk_firmware #24129: Error - Invalid keyboard level feature detected - tap_dance

digex, 2024-07-16:
### Issue Description

 qmk lint -kb handwired/digex/ortho_split_rgb_matrix_keyboard -km default

```
☒ handwired/digex/ortho_split_rgb_matrix_keyboard: Invalid keyboard level feature detected - tap_dance
☒ Lint check failed for: handwired/digex/ortho_split_rgb_matrix_keyboard
```

keyboard.json :

`"tap_dance" : true,`

my keymap

[`
https://pastebin.com/fnPaiBia
`](https://pastebin.com/fnPaiBia)

Otherwise, it compiles and works correctly.

fauxpark, 2024-07-16:
As the error message says, you cannot enable certain features at keyboard level. They will not work properly.

closed 2024-07-16 by fauxpark

digex, 2024-07-16:
I've been using this keymap for over three years now and everything has worked great. 
and now when lint I get this message. 
compilation takes place without errors and tap dance works correctly. 
I want to get rid of this message and add the keyboard to the repository

fauxpark, 2024-07-16:
Yes, this error message was introduced to prevent people from doing exactly this.

The Tap Dance feature requires the existence of an array named `tap_dance_actions` in your `keymap.c`. Because the Configurator does not build any keymap from this repo, but instead simply sends the JSON keymap to the backend, the compilation will fail if Tap Dance is enabled at keyboard level.