qmk/qmk_firmware #24167: Permission denied

Jark5455, 2024-07-21:
### Issue Description

Hello,

Currently I am trying to compile a custom firmware in msys, however no matter what I do I get the error.

```
☒ Unknown error attempting to load C:/Users/Me/qmk_firmware/keyboards/gmmk/pro/rev1/ansi:
        [Errno 13] Permission denied: 'C:/Users/Me/qmk_firmware/keyboards/gmmk/pro/rev1/ansi'
```

The `qmk_firmware` folder is owned by my user and my user has read and write access to that folder. I am not sure why qmk doesn't haver permission to access that folder.

fauxpark, 2024-07-21:
Impossible to tell without additional information and context. It could be solved by deleting and recloning, or it might simply be that your Windows install is broken in some weird way.

Jark5455, 2024-07-21:
Setting a default keyboard allows it to read, but without a default keyboard, I get the permission error.

closed 2024-07-21 by Jark5455

fauxpark, 2024-07-21:
I have no idea what that means. What command are you running?

Jark5455, 2024-07-21:
`qmk config user.keyboard=gmmk/pro/rev1/ansi`

fauxpark, 2024-07-22:
I can't reproduce this with the `default` keymap. The issue is something to do with your machine.