qmk/qmk_firmware #24175: How to read the 4-layer key? I want to press the key corresponding to the 4-layer output.

jeromeli002, 2024-07-22:
How to read the  4 layer key? I want to achieve that pressing the key corresponding to the 4th layer will output the key corresponding to the 5th layer, but the key that is pressed will output F11 instead of the 4th layer key. How should I write it?

[image]

filterpaper, 2024-07-26:
As noted, the `keycode` variable supplied to `process_record_user` will not change with `layer_*()` functions. However you can use another function to extract the keycode from your keymap using the `record` pointer, e.g.:
```c
tap_code16(keymap_key_to_keycode(4, record->event.key));
```
and
```c
tap_code16(keymap_key_to_keycode(5, record->event.key));
```

jeromeli002, 2024-07-29:
> As noted, the `keycode` variable supplied to `process_record_user` will not change with `layer_*()` functions. However you can use another function to extract the keycode from your keymap using the [...]

Thank you very much for your method, it works fine. I would like to ask how to rotate the PS canvas? I did not find any examples on qmk, nor did I get any methods on Google. But I saw others recommend that it can be supported. I can rotate by pressing R with a macro and then moving the mouse, but the effect looks unnatural.

filterpaper, 2024-07-29:
> I would like to ask how to rotate the PS canvas?

This is a QMK issue repository. Questions about PS is best directed at their support forums.

zvecr, 2024-08-03:
Closing as conversation has deviated away QMKs domain.

closed 2024-08-03 by zvecr