qmk/qmk_firmware #24190: [Bug] The firmware for Keychron Q2 Max Encoder Not Working

jaztsong, 2024-07-24:
### Describe the Bug

The compiled firmware simply just broke the keyboard. I used QMK ToolBox (GUI) to flash. It always works if I flash the official firmware (close-souced).

A close comparison between the `.bin` files of the firmware provided by the official website versus the qmk_firmware shows that the official bin file is 92KB whereas the qmk_firmware is only 48 KB. I reckon that some substantial components are missing. 

### Keyboard Used

Keychron/q2/ansi_encoder

### Operating System

Mac

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/<username>/Misc/qmk_firmware
Ψ Detected macOS 14.5 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.25.17
Ψ - Latest master: 2024-07-24 00:05:50 +0100 (60c5ad821c) -- Bump actions/labeler from 4 to 5 (#22608)
Ψ - Latest upstream/master: 2024-07-24 15:16:51 +1000 (8c35011d0a) -- [CLI] Only generate files if contents change. (#24038)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-07-24 00:05:50 +0100 (60c5ad821c) -- Bump actions/labeler from 4 to 5 (#22608)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 7.3
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

adophoxia, 2024-07-24:
You screwed up by flashing firmware for the Q2 when you have clearly stated you have a Q2 Max. You can pretty much figure out why that's the case given the lack of correlation between the two boards.

zvecr, 2024-07-24:
> **Keychron**
> Some Keychron boards are not merged into the main QMK repository yet. You should ask for help using their fork on the Keychron Discord server.
> 
> GitHub: https://github.com/Keychron/qmk_firmware/tree/playground
> Discord: https://discord.gg/6Sr9r9T7Nc

closed 2024-07-24 by zvecr

jaztsong, 2024-07-27:
Appreciate it. I will reach out to Keychron to find out a solution.

adophoxia, 2024-07-27:
@jaztsong, it is more easier to reach out over in their Discord for help regarding this.

jaztsong, 2024-07-27:
@adophoxia Yes, I'm on it now. Nice community.