qmk/qmk_firmware #24197: [Feature Request] Add support for a new keyboard

ThatOneOcto, 2024-07-25:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

Can you add the keychron c3 to upstream? Keychron themselves has made a fork for the c3 ([which can be found here](https://github.com/Keychron/qmk_firmware/tree/keychron_c3_pro)) but is outdaded, I would appreciate that considering the c1 and c2 are already upstream, thank you

daskygit, 2024-07-25:
Managed to click by mistake while reading on my phone, sorry.

daskygit, 2024-07-25:
While submitting a feature request you should have been presented with the following text.

> Please refrain from asking maintainers to add support for specific keyboards -- it is unlikely they will have hardware available, and will not be able to help.
Your best bet is to take the initiative, add support, then submit a PR yourself.

You could also ask Keychron to upstream the board but they're under no obligation to do so.

closed 2024-07-25 by daskygit