qmk/qmk_firmware #24204: [Feature Request] Mutually-exclusive key events a.k.a. Razer "Snap Tap" / Wooting "SOCD"

andrebrait, 2024-07-27:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

### Description

The newest fad in gaming keyboards seems to be the ability to define keys that are mutually exclusive so that games only react to your last pressed key, eliminating human error from the equation when it comes to techniques that require quickly alternating between keys, such as strife for correcting the aim / more quickly stopping movement in order to more quickly get the aim more precise in some first-person shooters.

It seems to be the most useful for switches that are capable of "analog" input, but I can see it being used for "digital" switches too.

Given this shouldn't be too hard to implement in QMK so that it can be reused by different boards, I thought of opening this feature request (which is more of an RFC, as I intend to implement it myself, if no one does it first).

Some open questions I already have in mind:

1. Should this be implemented at all? Some people think it's "cheating", but I disagree if that's a hardware feature not too dissimilar to a DPI toggle for a mouse or smoothing normally applied for accelerometers and analog sticks. I think that's also completely besides the point, given it's an optional feature, but I want other people's opinions. After all, I won't start working on something that won't be accepted anyway.
2. Does QMK also account for "analog" input keyboards (e.g. Hall Effect switches) at all nowadays? The implementation should be useful for both "analog" and "digital" switches.
3. Where should the implementation reside? I was thinking the usual approach of having a keyboard-level definition and a user-level definition of what keys are to be considered mutually-exclusive or a way to react and customize the processing of such thing + altering the generation of key events just after debouncing. It would obviously use some more memory (up to the size of the entire key matrix) to keep track and generate the right events, I suppose.

### References

1. https://www.razer.com/eu-en/technology/snap-tap-mode
2. https://wooting.io/post/socd-our-implementation-of-snap-tap
3. https://youtu.be/Feny5bs2JCg
4. https://youtu.be/RxEa7k8j1Ro

HorrorTroll, 2024-07-27:
Related to PR #24000

tzarc, 2024-07-27:
And analog input was always supported through custom matrix. You need to design the circuitry and write the matrix code yourself — there’s likely too much variance to allow for a core-side implementation.

andrebrait, 2024-07-27:
> Related to PR #24000

GitHub search apparently sucks, as that didn't show up at all before I opened this.

Definitely related, if not exactly the same thing.

I'll read the rest of the PR and tests later. For now, it isn't clear to me whether they function exactly the same as the "snap tap" aspect of it also comes from regenerating the press/release event when the corresponding mutual exclusive key gets pressed/release, while the original key keeps being pressed down (in other words: not requiring an actual reset for the key press to be sent again).

tzarc, 2024-07-27:
It’s phase 1 without it, phase 2 will have it. Different behaviour, different PRs. Already in the works by the same author.

tzarc, 2024-07-27:
I’ll also point out that the PR is locked because we’ve had upwards of 100 people in total demanding it. If you don’t mind I’m going to close this issue as it’s added noise for something already in the works.

closed 2024-07-27 by tzarc

andrebrait, 2024-07-27:
@tzarc absolutely fine. Hopefully this one gets indexed by Google/GitHub and pops up in search results

tzarc, 2024-07-27:
Thanks for being acommodating!

Geczy, 2024-08-26:
discussion locked again 2 weeks ago https://github.com/qmk/qmk_firmware/issues/24216

no movement still on this feature