qmk/qmk_firmware #24207: [Bug] Nix environment doesn't include qmk

Dietr1ch, 2024-07-28:
### Describe the Bug

The default environment won't work out of the box when relying only on shell.nix as `qmk` is not available.

This makes silly things like,
```fish
qmk setup
```

not work right away :(

### qmk doctor Output

```text
The program 'qmk' is not in your PATH. You can make it available in an
ephemeral shell by typing:
  nix-shell -p qmk
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I'm using `direnv` to load the right environment on each directory, but the one specified for qmk_firmware lacks the `qmk` command.

zvecr, 2024-07-29:
As you closed your "fix" as not required, I assume this should be closed too.

closed 2024-07-29 by zvecr

Dietr1ch, 2024-07-29:
Right