qmk/qmk_firmware #24262: [Feature Request] Implement ZMK's `require-prior-idle-ms` in QMK

amarakon, 2024-08-10:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

### Description

ZMK has the configuration option [require-prior-idle-ms](https://zmk.dev/docs/behaviors/hold-tap#advanced-configuration) for tap-hold keys. It effectively makes all tap-hold keys resolve as tapped when typing quickly. Aside from decreasing the possibility of accidentally triggering modifiers, it has the added benefit of eliminating the delay when typing quickly.

I did find a [pull request](https://github.com/manna-harbour/qmk_firmware/pull/56) on Manna Harbour's fork of the QMK firmware, but it's from two years ago and I can't merge it into my clone of this repository without a lot of merge conflicts. There's also the [achordion](https://getreuer.info/posts/keyboards/achordion/index.html) userspace library, but it can't get rid of the delay; it can only prevent modifiers from being triggered.

filterpaper, 2024-08-21:
It is a non-trivial request to implement this into QMK's core action tapping code. But you can easily do so in your user keymap code using the `pre_process_record_user` function with something like:
```c
#define IS_HOMEROW_MOD_TAP(kc) (              \
    IS_QK_MOD_TAP(kc)                      && \
    QK_MOD_TAP_GET_TAP_KEYCODE(kc) >= KC_A && \
    QK_MOD_TAP_GET_TAP_KEYCODE(kc) <= KC_Z    )

#define IS_TYPING(k) ( \
    ((uint8_t)(k) <= KC_Z || (uint8_t)(k) == KC_SPC) && \
    (last_input_activity_elapsed() < QUICK_TAP_TERM)    )

bool pre_process_record_user(uint16_t keycode, keyrecord_t *record) {
    static uint16_t prev_keycode, next_keycode;
    static bool     is_pressed[UINT8_MAX];
    static uint8_t  tap_keycode = QK_MOD_TAP_GET_TAP_KEYCODE(keycode);

    if (record->event.pressed) {
        // Cache previous and next input for tap-hold decisions
        prev_keycode = next_keycode;
        next_keycode = keycode;

        // Press the tap keycode if the tap-hold key follows the previous key swiftly
        if (IS_HOMEROW_MOD_TAP(keycode) && IS_TYPING(prev_keycode)) {
            is_pressed[tap_keycode] = true;
            record->keycode = tap_keycode;
        }
    }

    // Release the tap keycode if pressed
    else if (is_pressed[tap_keycode]) {
        is_pressed[tap_keycode] = false;
        record->keycode = tap_keycode;
    }

    return true;
}
```
The preprocessor macros can be adjusted to your preference. Unlike ZMK, you can decide on the preceding keycode that will trigger the feature. It uses the `keycode` container in the `keyrecord_t`. If you don't have `COMBO_ENABLE`, enable `REPEAT_KEY_ENABLE`.

amarakon, 2024-08-21:
First, thanks a lot for the code. I tested it and it works for home row mods. However, I also like to use my spacebar as a mod-tap key. This is how I set up my mod-taps:

- spacebar: mod-tap for gui
- s and l: mod-tap for alt
- d and k: mod-tap for shift
- f and j: mod-tap for control

I changed the `IS_HOMEROW_MOD_TAP` macro to the following:

```c
#define IS_HOMEROW_MOD_TAP(kc) ( \
    IS_QK_MOD_TAP(kc) && \
    ((QK_MOD_TAP_GET_TAP_KEYCODE(kc) >= KC_A && \
      QK_MOD_TAP_GET_TAP_KEYCODE(kc) <= KC_Z) || \
      QK_MOD_TAP_GET_TAP_KEYCODE(kc) == KC_SPC))
```

This makes it so that `KC_SPC` can also be considered a mod-tap key. It works for the most part, but sometimes it won't work and there'll be an unexpected delay when pressing space during a typing streak, as if I was using standard mod-taps.

I attached a video. I used [keyd](https://github.com/rvaiya/keyd) to monitor my keystrokes. By the way, Notice how there's a delay when typing the space right before the word `The`. Also notice how the difference in time between `space down` and `f down` is one millisecond. This probably means that the space is delayed until the F key is pressed. By the way, I'm using the Colemak layout so I press the F key to type the letter T.

https://github.com/user-attachments/assets/96b35bea-0999-4a32-a3da-edf997f1311e

amarakon, 2024-08-22:
I just realized that the reason I was experiencing that issue was because I also had a key-remapping daemon running which does the same thing. I was using it as a temporary solution for this problem and forgot to turn it off when I re-flashed my firmware. I'll still keep this issue open because I'd like to see this get implemented as a core QMK feature.

tzarc, 2025-01-25:
Someone would need to make a PR with associated unit tests.

getreuer, 2025-03-16:
I wrote a ["Tap Flow" community module](https://getreuer.info/posts/keyboards/tap-flow/index.html) for require-prior-idle behavior in QMK. The approach is based on and thanks to @filterpaper's https://github.com/qmk/qmk_firmware/issues/24262#issuecomment-2301722637 and excellent [Contextual Mod-Taps](https://github.com/filterpaper/qmk_userspace?tab=readme-ov-file#contextual-mod-taps) write up.

Tap Flow modifies mod-tap `MT` and layer-tap `LT` keys such that when pressed within a short timeout of the preceding key, the tapping behavior is triggered.

* Tap Flow acts only when a tap-hold press is within `TAP_FLOW_TERM` of the previous key event, which defaults to 150 ms. There are keycodes `TFLOW_U` / `TFLOW_D` that may be used to [tune the term dynamically](https://getreuer.info/posts/keyboards/tap-flow/index.html#tap_flow_term).
* An additional constraint, Tap Flow acts only when both the tap-hold key and the previous key are among Space, letters A–Z, and punctuations , `. ; /`. A [get_tap_flow() callback](https://getreuer.info/posts/keyboards/tap-flow/index.html#get_tap_flow) may be defined to customize this logic.

See the full documentation for how to install and further details:  
https://getreuer.info/posts/keyboards/tap-flow/index.html

filterpaper, 2025-03-17:
> I wrote a ["Tap Flow" community module](https://getreuer.info/posts/keyboards/tap-flow/index.html) for require-prior-idle behavior in QMK. The approach is based on and thanks to [@filterpaper](https://github.com/filterpaper)'s [#24262 (comment)](https://github.com/qmk/qmk_firmware/issues/24262#issuecomment-2301722637) and excellent [Contextual Mod-Taps](https://github.com/filterpaper/qmk_userspace?tab=readme-ov-file#contextual-mod-taps) write up.
> 
> Tap Flow modifies mod-tap `MT` and layer-tap `LT` keys such that when pressed within a short timeout of the preceding key, the tapping behavior is triggered.
> 
> * Tap Flow acts only when a tap-hold press is within `TAP_FLOW_TERM` of the previous key event, which defaults to 150 ms. There are keycodes `TFLOW_U` / `TFLOW_D` that may be used to [tune the term dynamically](https://getreuer.info/posts/keyboards/tap-flow/index.html#tap_flow_term).
> * An additional constraint, Tap Flow acts only when both the tap-hold key and the previous key are among Space, letters A–Z, and punctuations , `. ; /`. A [get_tap_flow() callback](https://getreuer.info/posts/keyboards/tap-flow/index.html#get_tap_flow) may be defined to customize this logic.
> 
> See the full documentation for how to install and further details: https://getreuer.info/posts/keyboards/tap-flow/index.html

It is a great addition to your list of modules @getreuer. I am curious about the choice of `uint8_t is_tapped[(MATRIX_ROWS * MATRIX_COLS + 7) / 8]` versus a `boolean` array of the calculated index, e.g.:
```c
static bool is_tapped[MATRIX_ROWS * MATRIX_COLS];
// Calculate matrix position index
const uint16_t array_index = (pos.row * MATRIX_COLS) + pos.col;
```
Number of ops in each matrix scan cycle will be reduced and tapped keys can be tracked with `is_tapped[array_index] = true / false`. Are there use-cases that I may have missed?

getreuer, 2025-03-17:
@filterpaper thank you for checking out the implementation!

Yes, that's a fair point. An array of `bool` would lower the CPU cost compared to the packed bit array. There's a cost tradeoff in space vs. CPU. I went with the packed bit array since the array is queried per event, and only on `MT` and `LT` keys, not per matrix scan. I would be hesitant about this if this was costing per matrix scan.

closed 2025-04-18 by amarz45

hitsmaxft, 2025-06-11:
@getreuer 

I have written custom code to improve flow tap with key record checking. Which did improve chordal holding on my split keyboard.

Custom function will check if  previous tap key record is on the oppsite hand, and   temporally disable flow tap checking while chordal inputting.

While fast type like ` o , hold d , c`, this mod will make it into hold , but not  `odc`

```
#ifdef FLOW_TAP_TERM

static float        border_for_splithand     = MATRIX_ROWS / 2.0 + 0.5;

static uint16_t     flow_tap_prev_keycode    = KC_NO;
static keyrecord_t *last_tap_flow_key_record = 0;

// ignore chordal tap flow
bool is_chordal_tap_flow(keyrecord_t *last, keyrecord_t *current) {
    return (last->event.key.row < border_for_splithand && current->event.key.row > border_for_splithand) || (last->event.key.row > border_for_splithand && current->event.key.row < border_for_splithand);
}
void flow_tap_chordal_reset(void) {

    last_tap_flow_key_record = 0;
    flow_tap_prev_keycode    = KC_NO;
}

// disable tap flow while pressing with two hand
uint16_t get_flow_tap_term(uint16_t keycode, keyrecord_t *record, uint16_t prev_keycode) {
    if (is_flow_tap_key(keycode) && is_flow_tap_key(prev_keycode)) {
        if (last_tap_flow_key_record != 0 && prev_keycode == flow_tap_prev_keycode && is_chordal_tap_flow(last_tap_flow_key_record, record)) {

#    ifdef TAP_FLOW_DEBUG
            dprintf("tap_flow disable by chordal typing");
#    endif // TAP_FLOW_DEBUG
            flow_tap_chordal_reset();
            return 0; // disable tap flow for chord hand or returning a smaller term 
        }
        // save last pressed key for next checking
        flow_tap_prev_time       = record->event.time;
        last_tap_flow_key_record = record;
        return FLOW_TAP_TERM;
    }
    flow_tap_chordal_reset();
    return 0;
}
```
These code isn't suit for common usage right now. I left it here for guys who need it.