qmk/qmk_firmware #24270: [Bug] custom firmware compiling for doio kb28 keeps giving errors with quantum/encoder.h and quantum/util.h

Nebula-0-System, 2024-08-12:
### Describe the Bug

i'm trying to compile custom firmware for the doio kb28 and i'm struggling as it's giving me constant trouble, namely with files in the quantum folder, which i feel shouldn't be happening, as those are core files that i feel i shouldn't have to edit to make this work. that said, i am having a hard time understanding what i'm looking at as this is my first try, so maybe i'm just doing something totally wrong.

the output i'm getting is as follows

`Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=kb28 KEYMAP=aleena KEYBOARD_FILESAFE=kb28 TARGET=kb28_aleena INTERMEDIATE_OUTPUT=.build/obj_kb28_aleena VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: .build/obj_kb28_aleena/src/default_keyboard.c                                           In file included from tmk_core/protocol/report.h:23:0,
                 from tmk_core/protocol/host.h:22,
                 from quantum/quantum.h:49,
                 from ./.build/obj_kb28_aleena/src/default_keyboard.h:27,
                 from .build/obj_kb28_aleena/src/default_keyboard.c:26:
./keyboards/kb28/config.h:33:30: error: ‘A5’ undeclared here (not in a function)
 #define ENCODERS_PAD_A { B5, A5, A3 }
                              ^
quantum/util.h:18:32: note: in definition of macro ‘MAX’
 #    define MAX(x, y) (((x) > (y)) ? (x) : (y))
                                ^
quantum/encoder.h:74:39: note: in expansion of macro ‘MAX’
 #    define NUM_ENCODERS_MAX_PER_SIDE MAX(NUM_ENCODERS_LEFT, NUM_ENCODERS_RIGHT)
                                       ^
quantum/util.h:47:54: note: in expansion of macro ‘IS_ARRAY’
 #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
                                                      ^
quantum/encoder.h:61:34: note: in expansion of macro ‘ARRAY_SIZE’
 #            define NUM_ENCODERS ARRAY_SIZE(((pin_t[])ENCODERS_PAD_A))
                                  ^
quantum/encoder.h:61:55: note: in expansion of macro ‘ENCODERS_PAD_A’
 #            define NUM_ENCODERS ARRAY_SIZE(((pin_t[])ENCODERS_PAD_A))
                                                       ^
quantum/encoder.h:63:35: note: in expansion of macro ‘NUM_ENCODERS’
 #        define NUM_ENCODERS_LEFT NUM_ENCODERS
                                   ^
quantum/encoder.h:74:43: note: in expansion of macro ‘NUM_ENCODERS_LEFT’
 #    define NUM_ENCODERS_MAX_PER_SIDE MAX(NUM_ENCODERS_LEFT, NUM_ENCODERS_RIGHT)
                                           ^
quantum/encoder.h:77:52: note: in expansion of macro ‘NUM_ENCODERS_MAX_PER_SIDE’
 #        define MAX_QUEUED_ENCODER_EVENTS MAX(4, ((NUM_ENCODERS_MAX_PER_SIDE) + 1))
                                                    ^
quantum/encoder.h:90:27: note: in expansion of macro ‘MAX_QUEUED_ENCODER_EVENTS’
     encoder_event_t queue[MAX_QUEUED_ENCODER_EVENTS];
                           ^
./keyboards/kb28/config.h:33:34: error: ‘A3’ undeclared here (not in a function)
 #define ENCODERS_PAD_A { B5, A5, A3 }
                                  ^
quantum/util.h:18:32: note: in definition of macro ‘MAX’
 #    define MAX(x, y) (((x) > (y)) ? (x) : (y))
                                ^
quantum/encoder.h:74:39: note: in expansion of macro ‘MAX’
 #    define NUM_ENCODERS_MAX_PER_SIDE MAX(NUM_ENCODERS_LEFT, NUM_ENCODERS_RIGHT)
                                       ^
quantum/util.h:47:54: note: in expansion of macro ‘IS_ARRAY’
 #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
                                                      ^
quantum/encoder.h:61:34: note: in expansion of macro ‘ARRAY_SIZE’
 #            define NUM_ENCODERS ARRAY_SIZE(((pin_t[])ENCODERS_PAD_A))
                                  ^
quantum/encoder.h:61:55: note: in expansion of macro ‘ENCODERS_PAD_A’
 #            define NUM_ENCODERS ARRAY_SIZE(((pin_t[])ENCODERS_PAD_A))
                                                       ^
quantum/encoder.h:63:35: note: in expansion of macro ‘NUM_ENCODERS’
 #        define NUM_ENCODERS_LEFT NUM_ENCODERS
                                   ^
quantum/encoder.h:74:43: note: in expansion of macro ‘NUM_ENCODERS_LEFT’
[... 458 lines cut ...]

#endif`

and my rules.mk in the root for the keyboard, and the keymap folder

`# Build Options
#   comment out to disable the options.
# for doio KB28
#
BOOTMAGIC_ENABLE = yes
MOUSEKEY_ENABLE = yes
EXTRAKEY_ENABLE = yes
CONSOLE_ENABLE = no
COMMAND_ENABLE = no
NKRO_ENABLE = yes
RGBLIGHT_ENABLE = no
AUDIO_ENABLE = no
OLED_ENABLE = yes
OLED_DRIVER = ssd1306
ENCODER_ENABLE = yes
UNICODE_ENABLE = yes
VIA_ENABLE = yes
RAW_ENABLE = yes`

i'm kinda hoping this is because i've overlooked something in the documentation, and not some issue with the actual firmware itself. i feel like maybe the code i was given is incompatible or maybe this is a much larger issue. i've been at this a while, so any and all help is appreciated as i've really hit a wall here.

Nebula-0-System, 2024-08-12:
> Pretty sure your `avr-gcc` is just plain too old.

if i may ask, what is that? and how do i update it?

tzarc, 2024-08-12:
It's the compiler that effectively converts the source code for QMK into a usable binary to flash your keyboard.
Unfortunately Debian-derived Linux distro's such as the Linux Mint you're using package an ancient copy of `avr-gcc`.

If you've got Docker installed, it's probably simplest to use that to build as the container includes a newer version:

```sh
./util/docker_cmd.sh qmk compile -kb kb28 -km aleena
```

Nebula-0-System, 2024-08-12:
> It's the compiler that effectively converts the source code for QMK into a usable binary to flash your keyboard. Unfortunately Debian-derived Linux distro's such as the Linux Mint you're using package an ancient copy of `avr-gcc`.
> 
> If you've got Docker installed, it's probably simplest to use that to build as the container includes a newer version:
> 
> ```shell
> ./util/docker_cmd.sh qmk compile -kb kb28 -km aleena
> ```

wow, had no idea it'd be something like that. i'll try it and get back to you

Nebula-0-System, 2024-08-12:
new errors, so at least we're getting somewhere. it prompted me to change some values in the code, but still giving errors now that i fixed that

```
Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=kb28 KEYMAP=aleena KEYBOARD_FILESAFE=kb28 TARGET=kb28_aleena INTERMEDIATE_OUTPUT=.build/obj_kb28_aleena VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

avr-gcc (GCC) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: .build/obj_kb28_aleena/src/default_keyboard.c                                            [OK]
Compiling: quantum/keymap_introspection.c                                                          quantum/keymap_introspection.c: In function 'keymap_layer_count_raw':
quantum/keymap_introspection.c:17:49: error: 'keymaps' undeclared (first use in this function)
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:20:12: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
     return NUM_KEYMAP_LAYERS_RAW;
            ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:17:49: note: each undeclared identifier is reported only once for each function it appears in
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:20:12: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
     return NUM_KEYMAP_LAYERS_RAW;
            ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: At top level:
quantum/keymap_introspection.c:17:49: error: 'keymaps' undeclared here (not in a function)
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:28:16: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
 _Static_assert(NUM_KEYMAP_LAYERS_RAW <= MAX_LAYER, "Number of keymap layers exceeds maximum set by DYNAMIC_KEYMAP_LAYER_COUNT");
                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:17:31: error: expression in static assertion is not an integer
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                               ^
quantum/keymap_introspection.c:28:16: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
 _Static_assert(NUM_KEYMAP_LAYERS_RAW <= MAX_LAYER, "Number of keymap layers exceeds maximum set by DYNAMIC_KEYMAP_LAYER_COUNT");
                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: In function 'keymap_layer_count_raw':
quantum/keymap_introspection.c:21:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake: *** [builddefs/common_rules.mk:373: .build/obj_kb28_aleena/quantum/keymap_introspection.o] Error 1
```

Nebula-0-System, 2024-08-12:
> If your `keyboards/doio/kb28/keymaps/default/keymap.c` is identical to `keymaps/doio/kb28/kb28g.c`, you effectively don't have a keymap (which is required).
> 
> Examining [other keyboards in the repository](https://github.com/search?q=repo%3Aqmk%2Fqmk_firmware%20path%3A%2Fkeyboards%5C%2F.%2B%5C%2Fkeymaps%5C%2Fdefault%5C%2Fkeymap.%2B%2F&type=code) along with [the documentation](https://docs.qmk.fm/keymap#keymap-overview) will provide helpful information about the structure of this file.

wow, i did not read that well enough, thought that was the default keymap, not that it was just for the LEDs. i also just realized the error specifically states "keymaps undeclared"

i'll fix that now and let you know.

Nebula-0-System, 2024-08-12:
ok got very far, all the way to flashing in fact (after removing `#include "kb28g.h"` in the keymap.c), but at the last line i got

```
sh: 1: hid_bootloader_cli: not found
gmake: *** [platforms/avr/flash.mk:173: flash] Error 127
```

which i checked, and it requires you to specify the microcontroller, which i did, and it still didn't work.

that said there was a previous issue i thought i solved where it wouldn't let me compile the right bootloader for the chip, so i just chose qmk-hid as it seemed like the generic option and it stopped giving me "invalid bootloader specified". maybe that was a mistake and i didn't actually solve the issue? i don't think so as it did seem to work afterwards.

Nebula-0-System, 2024-08-12:
> > so i just chose qmk-hid as it seemed like the generic option
> 
> There is no generic option for bootloaders; these are hardware-specific.
> 
> It would be a good idea to upload the original code as-provided to a public repository, since it is unknown what you've changed or what microcontroller the PCB uses. Given the original error message you mention of an _invalid_ bootloader, seeking assistance with the conversion in the [QMK Discord](https://discord.gg/Uq7gcHh) is also recommended.
> 
> It doesn't sound like the bootloader situation is simple/standard, and getting this wrong has the potential to "brick" the PCB. (You would need to acquire a hardware programmer to reprogram the bootloader using specific pins if that happens.)

alright, thank you. i'll move there and close this then as clearly it's not a bug or whatever, but has to do with the KB code. thanks for the help in getting me closer

closed 2024-08-12 by Nebula-0-System