qmk/qmk_firmware #24272: Entering Royal Kludge RK G68 bootloader

KrzysztofPrzygoda, 2024-08-13:
To @Thomaz-Peres (a [Royal Kludge RK G68 maintainer](https://github.com/qmk/qmk_firmware/blob/master/keyboards/rkg68/readme.md)) or anyone who can help.

Hi,

I'm struggling with entering in bootloader mode following [your guide here](https://github.com/qmk/qmk_firmware/blob/master/keyboards/rkg68/readme.md) on my RK G68 (837) with PCB 130-83703-12 2022/08/20 REV: 02.

**Bootmagic reset** and **Keycode in layout** methods doesn't work. So, I was trying to find physical button/pins on the back of PCB as instructed, but couldn't find anything like that. Can you provide any help with that?

The PCB bottom.
[image]

The PCB top.
[image]

The MCU.
[image]

fauxpark, 2024-08-13:
This board does not run QMK. BYK916 is an 8051 chip.

closed 2024-08-13 by fauxpark

KrzysztofPrzygoda, 2024-08-13:
> This board does not run QMK. BYK916 is an 8051 chip.

So, why is that keyboard in [repo](https://github.com/qmk/qmk_firmware/blob/master/keyboards/rkg68/readme.md)?

fauxpark, 2024-08-13:
It's not, at least, not the version you have. That one apparently runs on an Atmel ATmega32U4, which looks something like this:
[image]

KrzysztofPrzygoda, 2024-08-13:
RKG68 has never had Atmel MCU but SH68F90A / BYK916.

fauxpark, 2024-08-13:
Then I have no idea what #20138 is meant to be for... the image supplied shows a clearly wireless board. I'm thinking this needs to be removed.

Thomaz-Peres, 2024-08-13:
Yeah, I will do a PR to remove the file or update the Readme today.

We talked about that early in this [thread](https://github.com/Thomaz-Peres/qmk_firmware/commit/31e332e2ff6f40de2b13975fc96b7129b31bffa1#commitcomment-145306391)

And like I commented in the thread, sorry for the mistake, guys. Many people have problems because this PR

KrzysztofPrzygoda, 2024-08-13:
OK, thanks for clarification. It saved me some time.