qmk/qmk_firmware #24273: [Bug] `./util/docker_build.sh` should not require `docker-machine`

puchupala, 2024-08-13:
### Describe the Bug

I used to be able to `./util/docker_build.sh` on my Mac without having to install `docker-machine`. However, since https://github.com/qmk/qmk_firmware/pull/23988 make `./util/docker_build.sh` call `./util/docker_cmd.sh`, it now also (seemingly unnecessary) require `docker-machine` as well.

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-08-13:
You'll find that the `docker-machine` check has been present for much longer than the Pr you've quoted.

It's there to enable flashing -- historically we've had support requests as to why flashing doesn't work with docker builds, so I'd imagine this is why that check is present.

Feel free to raise a PR adding a new environment variable or similar mechanism to disable the check -- it'll need to stay on by default.

puchupala, 2024-08-14:
> You'll find that the `docker-machine` check has been present for much longer than the PR you've quoted.

@tzarc My apologies. I was using ZSA's fork on firmware23 branch, which doesn't have `docker-machine` check in `./util/docker_build.sh`.

> Feel free to raise a PR adding a new environment variable or similar mechanism to disable the check -- it'll need to stay on by default.

I'll get to that. Thank you very much for the suggestion.

closed 2024-08-14 by puchupala