qmk/qmk_firmware #24303: Removing layer related functionality

jkavalec, 2024-08-19:
### Issue Description

I am prototyping alternative layer solution, and I am wondering if it is possible to get rid of things related to the original layers.

As of now I am using layer functionality just to map keycodes like Key_1, Key_2 to the the first layer, then forgetting they exist:

```c
const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    [BASE] = LAYOUT_ergodox_80(
        _Key_00_, Key_01_, _Key_02_, _Key_03_, _Key_04_, _Key_05_, // and so on
```

I would prefer to get rid of this completely e.g. I don't want to have for instance global variable layer_state available which disallows mee to have var for my layer implementation to have some name. And I would also like this code not to increase firmware size.
 
Is there any way to do this? So that I can rely purely on matrix line/pos coordinates?

tzarc, 2024-08-19:
About the extent of it is to use `#define NO_ACTION_LAYER` and hope nothing breaks.
It's not really a tested configuration, so you're pretty much on your own.

jkavalec, 2024-08-19:
This was helpful, thank you very much.

I was able together with taps and oneshots get about 30% more space for my firmware. 
The only thing that still remains is that action_layer.h still adds layer_state definition like this "#    define layer_state 0" into resulting code so I still cannot use that variable name. This is small issue, and will most likely keep using layer_data name.

closed 2024-12-19 by github-actions[bot]