qmk/qmk_firmware #24362: [Bug] Macros are not working properly on Ubuntu 22.04 LTS

ckulemeyer, 2024-09-04:
### Describe the Bug

While this macro works fine on windows, it does not on ubuntu. For example, the following macro produces on winodows 
"|", | being the cursor, on ubuntu it produces either 2|2, "|2 or 2|". I tried different ways of writing the macro as indicated by the comments:

#include "sendstring_german.h"
  case QUOTES:
    if (record->event.pressed) {
      /* SEND_STRING("\"\""); */
      /* tap_code(KC_LEFT); */
      SEND_STRING(
                  /* SS_LSFT(SS_TAP(X_2) SS_TAP(X_2)) */
                  SS_LSFT("22")
                  SS_TAP(X_LEFT)
                  );
    }
    return false;

### Keyboard Used

lily58

### Operating System

ubuntu

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/chkuleme/qmk_firmware
Ψ Detected Linux (Ubuntu 22.04.4 LTS).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.26.0
Ψ - Latest master: 2024-09-04 03:57:19 +0100 (2b55e43e54) -- Bump peter-evans/create-pull-request from 6 to 7 (#24361)
Ψ - Latest upstream/master: 2024-09-04 03:57:19 +0100 (2b55e43e54) -- Bump peter-evans/create-pull-request from 6 to 7 (#24361)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-09-04 03:57:19 +0100 (2b55e43e54) -- Bump peter-evans/create-pull-request from 6 to 7 (#24361)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I am using a RP2040 Controller

fauxpark, 2024-09-04:
This is expected behaviour, if you have not matched your OS keyboard layout with the sendstring LUT header you are using.

closed 2024-09-04 by fauxpark

sigprof, 2024-09-04:
This actually looks more like the Wayland lost modifiers issue, which is claimed to be fixed upstream, but Ubuntu 22.04 LTS probably would never get that fix (maybe 24.04 has it though): https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/5890

As a workaround, you may try using `SEND_STRING_DELAY()` with an extra delay for your macros if you are using `"\""`, or adding explicit `SS_DELAY(x)` if you are using `SS_LSFT(SS_TAP(X_2))`.

ckulemeyer, 2024-09-04:
Sorry, I did not realize that you responded that fast. Thank you very much for your suggestions @sigprof. I will try both and maybe also switching back to x11.

ckulemeyer, 2024-09-04:
Thank you very much @sigprof, all my macros are now working on Ubuntu with wayland by simply changing the send_string function to send_string_delay(string, 10)