qmk/qmk_firmware #24407: [Bug] Swap-hands breaks OSL

ilikeheaps, 2024-09-18:
### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

ilikeheaps, 2024-09-18:
Perhaps a fix similar to https://github.com/qmk/qmk_firmware/issues/6943#issuecomment-1223723779 would work, I might try it if I find time.

sigprof, 2024-09-18:
`action.kind.id == ACT_SWAP_HANDS && action.swap.code == OP_SH_ONESHOT` should match `QK_SWAP_HANDS_ONE_SHOT` (`SH_OS`), not `SH_T(kc)`.  So that part of code keeps the one shot layer active only for `SH_OS`, and `SH_T(kc)` is not treated as a special case anywhere.

Maybe I should revive #9538, so that various user preferences in that area could be handled properly; one problem is that the API does not look good.

ilikeheaps, 2024-09-18:
Attempted a fix without understanding the code and guess what, seems like it broke something else. It might be more useful to me right now anyway.
```
diff --git a/quantum/action.c b/quantum/action.c
index 29822c39e9..3ad38b983a 100644
--- a/quantum/action.c
+++ b/quantum/action.c
@@ -378,7 +378,9 @@ void process_action(keyrecord_t *record, action_t action) {
 #    endif
                                               ))
 #    ifdef SWAP_HANDS_ENABLE
-        && !(action.kind.id == ACT_SWAP_HANDS && action.swap.code == OP_SH_ONESHOT)
+        && !(action.kind.id == ACT_SWAP_HANDS && (action.swap.code == OP_SH_ONESHOT
+                                                  || action.swap.code < 0xF0 // swap-hand tap key
+                                                  ))
 #    endif
         && keymap_config.oneshot_enable) {
         clear_oneshot_layer_state(ONESHOT_OTHER_KEY_PRESSED);
-- 
```

Now:
```
"r / 2"                                   -> r
OSM(MOD_LSFT) then OSL(1) then "r / 2"    -> @
SH_T(KC_SPC) +  "r / 2"                   -> i
SH_T(KC_SPC) +  OSL(1) + "r / 2"          -> 9
OSL(1) then SH_T(KC_SPC) +  "r / 2"       -> 9
toggle tap OSL(1); SH_T(KC_SPC) + "r / 2" -> 9

OSM(MOD_LSFT) then SH_T(KC_SPC) +  "r / 2"    -> I
OSM(MOD_LSFT) then quickly SH_T(KC_SPC) +  "r / 2" -> <SPC>, r ***expected I***
```

ilikeheaps, 2024-09-18:
> `action.kind.id == ACT_SWAP_HANDS && action.swap.code == OP_SH_ONESHOT` should match `QK_SWAP_HANDS_ONE_SHOT` (`SH_OS`), not `SH_T(kc)`. So that part of code keeps the one shot layer active only for `SH_OS`, and `SH_T(kc)` is not treated as a special case anywhere.

I suppose the issue with handling SH_T here is that we can't yet know whether it will become a modifier or a tap key?

sigprof, 2024-09-18:
`record->tap.count` is available in `process_action()` (already cached in `tap_count`), so you can determine whether the action is a tap or a hold and act accordingly (there is a check like that in the mod tap handling part above).

ilikeheaps, 2024-09-18:
Thanks, now tapping SH_T properly releases one shot layers/modifiers. I still have some issues, I guess due to delay in OSM (https://github.com/qmk/qmk_firmware/issues/1247 ?).

```
OSM(MOD_LSFT), wait ~0.5s, then SH_T(KC_SPC) +  "r / 2"    -> I
OSM(MOD_LSFT) then quickly SH_T(KC_SPC) +  "r / 2" -> <SPC>, r ***expected I***
```
(same issue with OSL)

sigprof, 2024-09-18:
That old issue looks unrelated, and from your report it seems that in the `SH_T(KC_SPC) +  "r / 2"` your `SH_T(KC_SPC)` press gets handled like a tap instead of a hold — so either you need to do some tuning for your tap/hold decision modes, or maybe the tap detection gets disturbed by the previous `OSM(MOD_LSFT)` press (may it be the case that `OSM(MOD_LSFT)` is not yet released before `SH_T(KC_SPC)` is pressed?).

ilikeheaps, 2024-09-18:
My wild guess came from timing weirdly matching OSM tapping timeout. I tried producing some logs and surprisingly got 4 different results for the same `OSM(MOD_LSFT) then SH_T(KC_SPC) +  "r / 2"` sequence. Haven't parsed them yet, some might turn out to be user error. I'm pretty sure I released `OSM(MOD_LSFT)` first as I'm using the same finger for them.

"I"
```
2024-09-18 23:13:36 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:13:36 ilikeheaps:plyboard:1: EVENT: 0304d(57227)
2024-09-18 23:13:36 ilikeheaps:plyboard:1: Tapping: Start(Press tap key).
2024-09-18 23:13:36 ilikeheaps:plyboard:1: TAPPING_KEY=0304d(57227):0 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: processed: 0304d(57227):0 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:13:36 ilikeheaps:plyboard:1: EVENT: 0304u(57326)
2024-09-18 23:13:36 ilikeheaps:plyboard:1: Tapping: First tap(0->1).
2024-09-18 23:13:36 ilikeheaps:plyboard:1: TAPPING_KEY=0304d(57227):1 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: ACTION: ACT_LMODS_TAP[2:00] layer_state:  default_layer_state: 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: MODS_TAP: Oneshot: start
2024-09-18 23:13:36 ilikeheaps:plyboard:1: waiting_buffer_enq: { [0]=0304u(57326):1  }
2024-09-18 23:13:36 ilikeheaps:plyboard:1: ---- action_exec: process waiting_buffer -----
2024-09-18 23:13:36 ilikeheaps:plyboard:1: Tapping: Tap release(1)
2024-09-18 23:13:36 ilikeheaps:plyboard:1: ACTION: ACT_LMODS_TAP[2:00] layer_state:  default_layer_state: 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: TAPPING_KEY=0304u(57326):1 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: processed: waiting_buffer[0] =0304u(57326):1 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: 
2024-09-18 23:13:36 ilikeheaps:plyboard:1: Tapping: End(Timeout after releasing last tap): 0000u(57526)
2024-09-18 23:13:36 ilikeheaps:plyboard:1: TAPPING_KEY=0000u(0):0 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:13:37 ilikeheaps:plyboard:1: EVENT: 0305d(57779)
2024-09-18 23:13:37 ilikeheaps:plyboard:1: Tapping: Start(Press tap key).
2024-09-18 23:13:37 ilikeheaps:plyboard:1: TAPPING_KEY=0305d(57779):0 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: processed: 0305d(57779):0 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:13:37 ilikeheaps:plyboard:1: EVENT: 0102d(57942)
2024-09-18 23:13:37 ilikeheaps:plyboard:1: waiting_buffer_enq: { [1]=0602d(57942):0  }
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ---- action_exec: process waiting_buffer -----
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: Tapping: End. Timeout. Not tap(0): 0000u(57979)
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ACTION: ACT_SWAP_HANDS[0:2C] layer_state:  default_layer_state: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: TAPPING_KEY=0000u(0):0 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ACTION: ACT_LMODS[0:0F] layer_state:  default_layer_state: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: keyboard_report: 02 | 0F 00 00 00 00 00 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: processed: waiting_buffer[1] =0602d(57942):0 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:13:37 ilikeheaps:plyboard:1: EVENT: 0305u(58017)
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ACTION: ACT_SWAP_HANDS[0:2C] layer_state:  default_layer_state: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: processed: 0305u(58017):0 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:13:37 ilikeheaps:plyboard:1: EVENT: 0102u(58041)
2024-09-18 23:13:37 ilikeheaps:plyboard:1: ACTION: ACT_LMODS[0:0F] layer_state:  default_layer_state: 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: keyboard_report: 00 | 00 00 00 00 00 00 
2024-09-18 23:13:37 ilikeheaps:plyboard:1: processed: 0602u(58041):0 
```

"R"
```
2024-09-18 23:14:23 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:14:23 ilikeheaps:plyboard:1: EVENT: 0304d(38558)
2024-09-18 23:14:23 ilikeheaps:plyboard:1: Tapping: Start(Press tap key).
2024-09-18 23:14:23 ilikeheaps:plyboard:1: TAPPING_KEY=0304d(38558):0 
2024-09-18 23:14:23 ilikeheaps:plyboard:1: processed: 0304d(38558):0 
2024-09-18 23:14:23 ilikeheaps:plyboard:1: 
2024-09-18 23:14:23 ilikeheaps:plyboard:1: 
[... 154 lines cut ...]
2024-09-18 23:18:05 ilikeheaps:plyboard:1: Tapping: First tap(0->1).
2024-09-18 23:18:05 ilikeheaps:plyboard:1: TAPPING_KEY=0305d(62768):1-
2024-09-18 23:18:05 ilikeheaps:plyboard:1: ACTION: ACT_SWAP_HANDS[0:2C] layer_state:  default_layer_state: 
2024-09-18 23:18:05 ilikeheaps:plyboard:1: keyboard_report: 02 | 2C 00 00 00 00 00 
2024-09-18 23:18:05 ilikeheaps:plyboard:1: waiting_buffer_enq: { [3]=0602d(62861):0  [4]=0305u(62933):1- }
2024-09-18 23:18:05 ilikeheaps:plyboard:1: ---- action_exec: process waiting_buffer -----
2024-09-18 23:18:05 ilikeheaps:plyboard:1: Tapping: key event while last tap(>0).
2024-09-18 23:18:05 ilikeheaps:plyboard:1: ACTION: ACT_LMODS[0:0F] layer_state:  default_layer_state: 
[... cut ...]
2024-09-18 23:18:05 ilikeheaps:plyboard:1: 
2024-09-18 23:18:05 ilikeheaps:plyboard:1: 
2024-09-18 23:18:05 ilikeheaps:plyboard:1: 
2024-09-18 23:18:05 ilikeheaps:plyboard:1: ---- action_exec: start -----
2024-09-18 23:18:05 ilikeheaps:plyboard:1: EVENT: 0102u(62969)
2024-09-18 23:18:05 ilikeheaps:plyboard:1: ACTION: ACT_LMODS[0:0F] layer_state:  default_layer_state: 
2024-09-18 23:18:05 ilikeheaps:plyboard:1: keyboard_report: 00 | 00 00 00 00 00 00 
2024-09-18 23:18:05 ilikeheaps:plyboard:1: processed: 0602u(62969):0 
```

ilikeheaps, 2024-09-19:
While analyzing the logs I found out (a part of?) the issue happens on master regardless of one shot feature: `SH_T(KC_SPC) +  "r / 2"` produces `i` or ` i` depending on timing. Posted that in https://github.com/qmk/qmk_firmware/issues/24410

ilikeheaps, 2024-09-26:
Code I wrote above along with a (one-line) fix mentioned in https://github.com/qmk/qmk_firmware/pull/24420 resolved my issues (there remains an issue with PERMISSIVE_HOLD, also in that draft, but I'm not using that mode anymore). If someone wants to check the changes I would tidy them up (and I could substantiate how it increases the utility of OSM and SH for one handed use), otherwise this may be closed.