qmk/qmk_firmware #24417: [Bug] input_club/ergodox_infinity unresponsive on latest version

etdub, 2024-09-20:
### Describe the Bug

input_club/ergodox_infinity doesn't work when compiling against HEAD. Display lights up but doesn't render anything. No output when connected. 

QMK works when compiling from cd0043de8f (0.24.7)

### Keyboard Used

input_club/ergodox_infinity

### Operating System

Windows

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: C:/Users/ericw/qmk_firmware
Ψ Detected Windows 10 (10.0.22631).
Ψ QMK MSYS version: 1.9.0
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.26.4
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2024-09-20 02:28:57 +0100 (61778f6136) -- Fix a_jazz/akc084 render (#24411)
Ψ - Latest upstream/master: 2024-09-20 02:28:57 +0100 (61778f6136) -- Fix a_jazz/akc084 render (#24411)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-09-20 02:28:57 +0100 (61778f6136) -- Fix a_jazz/akc084 render (#24411)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.2.0
Ψ Found avr-gcc version 12.2.0
Ψ Found avrdude version 7.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Would you like to clone the submodules? [Y/n] y
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-09-23:
You'll have to `git bisect` on your own I'm afraid -- none of the maintainers have access to this board.

etdub, 2024-09-23:
Understandable. I'll see if I can find the commit.

sigprof, 2024-09-23:
Looks like the ST7565 LCD driver calls `spi_start()` without `spi_stop()` in [`st7565_render()`](https://github.com/qmk/qmk_firmware/blob/master/drivers/lcd/st7565.c#L182-L188).  Adding a `spi_stop();` line before `st7565_on();` might fix things (`st7565_on()` may call `spi_start()` again, so the `spi_stop()` call needs to be done before).

(Apparently `input_club/ergodox_infinity` is the only user of that ST7565 driver.)

etdub, 2024-09-23:
It looks like adding a `spi_stop()` did the trick. Able to compile from HEAD and keyboard seems to be working.

Appreciate the help especially when this is the only keyboard using this driver!

PR #24429 (2024-09-26): input_club/ergodox_infinity ST7565 call spi_stop
Fixes #24417

## Description

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* #24417

## Checklist

- [ ] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist) and have mad [... cut]
closed 2024-09-26 by drashna