qmk/qmk_firmware #24455: [Bug] Mouse Button presses KC_BTN* only work when mouse cursor is moving

btittelbach, 2024-10-01:
### Describe the Bug

## Expected behaviour 
- If the mouse cursor is resting, e.g. the trackpoint is not being touched, the mouse buttons should work

## Actual behaviour
- The mouse buttons don't work, unless the mouse cursor is also being moved

## Blamed commit

I managed to trace the bug to the following commit: a850f7d69509a99ef46dd2790c2e55171db16e7a
Mouse buttons work right before the commit was applied and stopped working with this commit.

## keyboard configuration

```
ENCODER_MAP_ENABLE = yes
WS2812_DRIVER = vendor
RGBLIGHT_ENABLE = yes
COMBO_ENABLE = yes
```

```json
{
.....
   "processor": "RP2040",                               
    "bootloader": "rp2040",                              
    "features": {                                        
        "bootmagic": true,                                                           
        "encoder": true,                               
        "extrakey": true,                                         
        "mousekey": true,                                            
        "ps2mouse": true                                        
    },                                                           
    "ps2": {                                                        
        "clock_pin": "GP27",                                          
        "data_pin": "GP26",                             
        "driver": "vendor",                              
        "enabled": true,                                 
        "mouse_enabled": true                            
    },
...
}                            
```

### Keyboard Used

qmk compile -kb chrumm/leds38 -km btittelbach_65keys_mouse_rgblight

### Link to product page (if applicable)

https://github.com/btittelbach/chrumm-qmk/tree/rgblight

### Operating System

GNU/Linux

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

btittelbach, 2024-10-01:
@strobo5 maybe you could take a look?

PeterMortensen, 2024-10-01:
What [PS/2](https://en.wikipedia.org/wiki/PS/2_port) device was it tested with?

A [Trackpoint from a Thinkpad T61](https://github.com/qmk/qmk_firmware/pull/23694)? An [Azoteq ProxSense TPS65](https://www.reddit.com/r/ErgoMechKeyboards/comments/10gnhm5/azoteq_touchpad_with_pico_bigger_but_cheaper_than/)? Something else?

strobo5, 2024-10-01:
Strange. I have the same question as PeterMortensen. Looking at the code, it's not immediately obvious to me what could be going on..

For #22265 / #23694 I used `qmk console` to get debugging output from the PS/2 driver. Maybe you can do the same, paste some of the output here, and we can try to make sense of it together.

These are my **possibly outdated** notes on how to get this log (check the documentation for reference):

In the `info.json`, set `"console": true`.

In the `keymap.c` add this to `keyboard_post_init_user()`:
```
debug_enable=true;
debug_mouse=true;
```
In the `config.h` add this:
```
#define PS2_MOUSE_DEBUG_RAW
#define PS2_MOUSE_DEBUG_HID
```

Compile a new image, then check `dmesg` for the hidraw number, e.g.:
```
hid-generic 0003:FEED:0000.001D: hiddev98,hidraw6: USB HID v1.11 Device
```
Then `chmod 666 /dev/hidraw6` (as root / with sudo).

Now use `qmk console`.

btittelbach, 2024-10-02:
> What [PS/2](https://en.wikipedia.org/wiki/PS/2_port) device was it tested with?
> 

I am not sure this is related to the PS/2 device at all, since the mouse keys are normal keys on keyboard on layer 1. I hold OSL(1) and then have access to KC_BTN1 to e.g. double-click. The buttons are not, in hardware, related to the PS/2 input at all.

Nevertheless:

This is one of my keyboards with the problem:
[image]

It used a _surnaiee MS8006_ China Trackpoint USB Evaluation Module and de-soldered the blue trackpoint module from the USB-Module. (see below) The blue module just outputs PS/2 on two wires, which I wired to the RP2040. The module is actually very good and precise.

[image]

sigprof, 2024-10-02:
I suppose the following happens here:
- `ps2_mouse_task()` gets called on every pass of the QMK main loop, which happens more frequently than the reports coming from the PS/2 device, therefore sometimes `ps2_mouse_task()` does not get any data from the PS/2 device.
- Before #23694 `ps2_mouse_task()` treated the case when there is no PS/2 report as if the PS/2 device returned a report containing all zero values (because the call to `ps2_mouse_clear_report()` on the previous iteration had it set that way). This caused problems when using a PS/2 device with its own set of mouse buttons, because the button state alternated between the state actually reported by the PS/2 device and no buttons pressed.
- After #23694 `ps2_mouse_task()` just does nothing if there is no PS/2 report; however, by doing nothing at all it also ignores the button state passed through the `tp_buttons` variable, and `tp_buttons` is where `register_mouse()` places the button state updated by keycodes like `KC_BTN1` when the PS/2 mouse support is enabled (the `mousekeys_send()` call gets skipped in this configuration, because it is intended that the PS/2 code would send the mouse report). So if the PS/2 device does not send any reports (e.g., because there is no movement and no actual buttons on the PS/2 device), `register_mouse()` updates only the `tp_buttons` variable, but `ps2_mouse_task()` ignores those updates and does not send the mouse report.

The proper way to fix all this is probably to make the PS/2 mouse support use the pointing device code instead of sending the USB reports itself.  A quicker fix might be to revert the changes done in #23694, and then remove `mouse_report->buttons = 0;` from `ps2_mouse_clear_report()`; then `ps2_mouse_task()` would treat the “no PS/2 report” case as “no movement, button state unchanged from the previously reported one”, and then the subsequent code would combine that button state with `tp_buttons` and send the mouse report if something had changed.

strobo5, 2024-10-07:
Thank you for these explanations, I had no idea about this mechanism with `register_mouse()`. I vaguely remember being confused by the `tp_buttons` variable.

> The proper way to fix all this is probably to make the PS/2 mouse support use the pointing device code instead of sending the USB reports itself. A quicker fix might be to revert the changes done in [...]

I tried the quick fix with my T61 keyboard (on top of #23696 which is still not merged) and did not notice any erratic behavior. It seems to fix #22265 just like #23694 did.

tachikomachann, 2024-11-17:
Hi, I have a similar issue #24599  but your quick fix() seems not working..  @sigprof @strobo5 
1. revert the changes done in #23694 
2. remove mouse_report->buttons = 0 from ps2_mouse_clear_report();

HattDroid, 2025-06-17:
After testing around a few settings related to PS/2, this is the setting that finally allowed mouse clicks from the keyboards button when using a PS/2 trackpoint without buttons:

`#define PS2_MOUSE_USE_REMOTE_MODE
`

The only problem I have now is switching to remote mode instead of stream mode is extremely laggy for most of the time which makes typing impossible.

Would anyone have an explanation of why remote mode actually works for mouse clicks now? And how would it be possible to get those clicks to work in stream mode?

Jeff-WuYo, 2025-07-20:
@HattDroid I don't experience the lag with basic trackpoint config (with `#define PS2_MOUSE_USE_REMOTE_MODE` ), however, if I enable scroll button related config, it is super laggy. I couldn't input anything. I also rotated trackpoint.

btittelbach, 2026-01-17:
tested with 0.31.8 and verified that this is NOT fixed

tested with https://github.com/qmk/qmk_firmware/pull/22532 aka joh:pointing_device_ps2_mouse and can verify that this fixes the bug
(though default mouse directions seems to be inverted, on rp2040 at least)