qmk/qmk_firmware #24470: [Bug] problem with backlight pwm (RP2040)

Gabriele-tomai00, 2024-10-10:
### Describe the Bug

Hi, I'd like to create a simple backlight with a led no RGB and no dimmer (or better yet, not necessary).
I'm using a raspberry pi pico RP2040 and when I add the rule:  ```BACKLIGHT_DRIVER = pwm```
I get this error during compilation (I also specified the pin in config.h:  ```#define BACKLIGHT_PIN GP23```):
```
Compiling: platforms/chibios/drivers/backlight_pwm.c                                               platforms/chibios/drivers/backlight_pwm.c:51:8: error: unknown type name 'PWMConfig'
 static PWMConfig pwmCFG = {
        ^~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:52:5: error: field name not in record or union initializer
     .frequency = BACKLIGHT_PWM_COUNTER_FREQUENCY, /* PWM clock frequency  */
     ^
platforms/chibios/drivers/backlight_pwm.c:52:5: note: (near initialization for 'pwmCFG')
platforms/chibios/drivers/backlight_pwm.c:53:5: error: field name not in record or union initializer
     .period    = BACKLIGHT_PWM_PERIOD,            /* PWM period in counter ticks. e.g. clock frequency is 10KHz, period is 256 ticks then t_period is 25.6ms */
     ^
platforms/chibios/drivers/backlight_pwm.c:53:5: note: (near initialization for 'pwmCFG')
In file included from platforms/chibios/_wait.h:20,
                 from platforms/wait.h:25,
                 from platforms/chibios/drivers/backlight_pwm.c:3:
platforms/chibios/chibios_config.h:39:45: error: excess elements in scalar initializer [-Werror]
 #    define BACKLIGHT_PWM_COUNTER_FREQUENCY 1000000
                                             ^~~~~~~
platforms/chibios/chibios_config.h:41:38: note: in expansion of macro 'BACKLIGHT_PWM_COUNTER_FREQUENCY'
 #        define BACKLIGHT_PWM_PERIOD BACKLIGHT_PWM_COUNTER_FREQUENCY / 2048
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:53:18: note: in expansion of macro 'BACKLIGHT_PWM_PERIOD'
     .period    = BACKLIGHT_PWM_PERIOD,            /* PWM period in counter ticks. e.g. clock frequency is 10KHz, period is 256 ticks then t_period is 25.6ms */
                  ^~~~~~~~~~~~~~~~~~~~
platforms/chibios/chibios_config.h:39:45: note: (near initialization for 'pwmCFG')
 #    define BACKLIGHT_PWM_COUNTER_FREQUENCY 1000000
                                             ^~~~~~~
platforms/chibios/chibios_config.h:41:38: note: in expansion of macro 'BACKLIGHT_PWM_COUNTER_FREQUENCY'
 #        define BACKLIGHT_PWM_PERIOD BACKLIGHT_PWM_COUNTER_FREQUENCY / 2048
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:53:18: note: in expansion of macro 'BACKLIGHT_PWM_PERIOD'
     .period    = BACKLIGHT_PWM_PERIOD,            /* PWM period in counter ticks. e.g. clock frequency is 10KHz, period is 256 ticks then t_period is 25.6ms */
                  ^~~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c: In function 'backlight_init_ports':
platforms/chibios/drivers/backlight_pwm.c:89:11: error: request for member 'channels' in something not a structure or union
     pwmCFG.channels[BACKLIGHT_PWM_CHANNEL - 1].mode = PWM_OUTPUT_MODE;
           ^
platforms/chibios/drivers/backlight_pwm.c:37:33: error: 'PWM_OUTPUT_ACTIVE_HIGH' undeclared (first use in this function); did you mean 'PWM_OUTPUT_MODE'?
 #        define PWM_OUTPUT_MODE PWM_OUTPUT_ACTIVE_HIGH;
                                 ^~~~~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:89:55: note: in expansion of macro 'PWM_OUTPUT_MODE'
     pwmCFG.channels[BACKLIGHT_PWM_CHANNEL - 1].mode = PWM_OUTPUT_MODE;
                                                       ^~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:37:33: note: each undeclared identifier is reported only once for each function it appears in
 #        define PWM_OUTPUT_MODE PWM_OUTPUT_ACTIVE_HIGH;
                                 ^~~~~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:89:55: note: in expansion of macro 'PWM_OUTPUT_MODE'
     pwmCFG.channels[BACKLIGHT_PWM_CHANNEL - 1].mode = PWM_OUTPUT_MODE;
                                                       ^~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:90:5: error: implicit declaration of function 'pwmStart'; did you mean 'usbStart'? [-Werror=implicit-function-declaration]
     pwmStart(&BACKLIGHT_PWM_DRIVER, &pwmCFG);
     ^~~~~~~~
     usbStart
platforms/chibios/drivers/backlight_pwm.c:22:34: error: 'PWMD4' undeclared (first use in this function); did you mean 'PWM'?
 #    define BACKLIGHT_PWM_DRIVER PWMD4
                                  ^~~~~
platforms/chibios/drivers/backlight_pwm.c:90:15: note: in expansion of macro 'BACKLIGHT_PWM_DRIVER'
     pwmStart(&BACKLIGHT_PWM_DRIVER, &pwmCFG);
               ^~~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c: In function 'backlight_set':
platforms/chibios/drivers/backlight_pwm.c:109:9: error: implicit declaration of function 'pwmDisableChannel' [-Werror=implicit-function-declaration]
         pwmDisableChannel(&BACKLIGHT_PWM_DRIVER, BACKLIGHT_PWM_CHANNEL - 1);
         ^~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:22:34: error: 'PWMD4' undeclared (first use in this function); did you mean 'PWM'?
 #    define BACKLIGHT_PWM_DRIVER PWMD4
                                  ^~~~~
platforms/chibios/drivers/backlight_pwm.c:109:28: note: in expansion of macro 'BACKLIGHT_PWM_DRIVER'
         pwmDisableChannel(&BACKLIGHT_PWM_DRIVER, BACKLIGHT_PWM_CHANNEL - 1);
                            ^~~~~~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:113:9: error: implicit declaration of function 'pwmEnableChannel' [-Werror=implicit-function-declaration]
         pwmEnableChannel(&BACKLIGHT_PWM_DRIVER, BACKLIGHT_PWM_CHANNEL - 1, PWM_FRACTION_TO_WIDTH(&BACKLIGHT_PWM_DRIVER, 0xFFFF, duty));
         ^~~~~~~~~~~~~~~~
platforms/chibios/drivers/backlight_pwm.c:113:76: error: implicit declaration of function 'PWM_FRACTION_TO_WIDTH' [-Werror=implicit-function-declaration]
         pwmEnableChannel(&BACKLIGHT_PWM_DRIVER, BACKLIGHT_PWM_CHANNEL - 1, PWM_FRACTION_TO_WIDTH(&BACKLIGHT_PWM_DRIVER, 0xFFFF, duty));
                                                                            ^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
gmake: *** [builddefs/common_rules.mk:376: .build/obj_split_keyboard_gabriele_vial/platforms/chibios/drivers/backlight_pwm.o] Error 1
```

I soled the problem changing che rule with ```ACKLIGHT_DRIVER = software```. 
Actually I'd like only a combination to turn on/off the light, but I don't understeand this problem. Anyone can help me?

Thank you so much

### Keyboard Used

personal

### Operating System

macos 15.0.1

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/gabriele/vial-qmk
Ψ Detected macOS 15.0.1 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: vial
Ψ Repo version: 0.25.9
Ψ - Latest vial: 2024-09-07 20:23:41 -0500 (bb7961c3c2) -- Ported magic_force/mf34 to VIAL (#780)
Ψ - Latest upstream/master: 2024-09-28 01:58:22 +0100 (4d6337539d) -- Revert "Bump JamesIves/github-pages-deploy-action from 4.6.4 to 4.6.6" (#24450)
Ψ - Latest upstream/develop: 2024-09-28 00:58:56 +0000 (1118821178) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2024-06-24 12:29:30 +1000 (03e688e91f) -- Add support for userspace to docker build commands. (#23988)
Ψ - Common ancestor with upstream/develop: 2024-06-24 12:29:30 +1000 (03e688e91f) -- Add support for userspace to docker build commands. (#23988)
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [X] Karabiner (macOS)

Gabriele-tomai00, 2024-10-15:
with software pwm the brightness is too low I don't know why. I also tried adding another mosfet in parallel but it is always low (so it is not the mosfet's fault). If I connect the led directly the brightness is much higher

ale1800, 2025-03-02:
Hi, can you explain how you solved that? I want to control some leds using pwm but I receivce the same errors during thecompilation.

Gabriele-tomai00, 2025-03-04:
Hi, if I remember correctly, you need to add the files mcuconf.h and halconf.h to define which pwm to use.
This is my mcuconf.h:
```
#pragma once

#include_next <mcuconf.h>

#undef RP_PWM_USE_PWM5
#define RP_PWM_USE_PWM5 TRUE
```
And halconf.h
```
#pragma once

#define HAL_USE_PWM TRUE

#include_next <halconf.h>
```
And in config.h
```
#define BACKLIGHT_PWM_DRIVER PWMD5
#define BACKLIGHT_PWM_CHANNEL RP2040_PWM_CHANNEL_A
```
The pwm channel depends to what GPIO you are going to use. Check che datasheet of the rapsberry rp2040

ale1800, 2025-03-04:
Great, thanks!