qmk/qmk_firmware #24489: PILA87 Firmware Flashing - Component Location

rango886, 2024-10-14:
### Issue Description

I recently purchased a PILA87 kit from Taobao and am preparing to flash the firmware. While reviewing the readme file at https://github.com/qmk/qmk_firmware/tree/master/keyboards/phage_studio/pila87, I noticed instructions regarding two electronic components that need to be soldered to the circuit board for flashing.
Could you please clarify the location of these components on the PILA87 PCB? A diagram or photo highlighting their positions would be incredibly helpful.
Thank you for your time and assistance.

HorrorTroll, 2024-10-14:
[image]
Here you go, crystal is 4 pad and other 2 is capacitor.

rango886, 2024-10-15:
> ![IMG_8558](https://private-user-images.githubusercontent.com/24840279/376195175-e209283e-58ce-449a-b39d-c0ef22563aec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg5NTU2MjQsIm5iZiI6MTcyODk1NTMyNCwicGF0aCI6Ii8yNDg0MDI3OS8zNzYxOTUxNzUtZTIwOTI4M2UtNThjZS00NDlhLWIzOWQtYzBlZjIyNTYzYWVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE1VDAxMjIwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI2MjMxMzU2YmNkYTFlODM1NTRmODcwZDk3ZDZjY2JkMmMzODY1ZjFkODA0YjQxY2RmYzdmYmIyYTRmOTZiMDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.x-yzBFhKWqXy4TdpU7-girgS6twDkhAfQFK5r0E33F8) Here you go, crystal is 4 pad and other 2 is capacitor. Try using multimeter to check which pad on crystal is OSC_IN and OSC_OUT before soldering.

Thank you very much, and I've noticed that firmware flashing with an ST-Link V2 utilizes a four-wire connection. Could you please clarify the proper connection of these four wires between the ST-Link V2 and the target PCB? 

[image]
[image]

HorrorTroll, 2024-10-15:
> Thank you very much, and I've noticed that firmware flashing with an ST-Link V2 utilizes a four-wire connection. Could you please clarify the proper connection of these four wires between the ST-Link V2 and the target PCB?

Here the details:
- 1: N/A
- 2: N/A
- 3: GND
- 4: SWDIO
- 5: SWCLK
- 6: 3.3V (or 3V3)

closed 2025-02-13 by github-actions[bot]