qmk/qmk_firmware #24492: [Bug] Unicode Map can't output Cyrillic characters with code 0x##0

kdlrose, 2024-10-14:
### Describe the Bug

Firmware compiles and flashes with no issues in QMK CLI, but on use these characters don't actually appear when typed:

- А: 0x0410
- Р: 0x0420
- а: 0x0430
- р: 0x0440

### Keyboard Used

planck/rev7

### Operating System

macOS 12

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /Users/sailorfe/s/qmk_firmware
Ψ Detected macOS 12.7.6 (Intel).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.26.8
Ψ - Latest master: 2024-10-12 00:29:32 +0100 (11daef82e0) -- zsa/moonlander: Fix execution of user callback (#24475)
Ψ - Latest upstream/master: 2024-10-12 00:29:32 +0100 (11daef82e0) -- zsa/moonlander: Fix execution of user callback (#24475)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-10-12 00:29:32 +0100 (11daef82e0) -- zsa/moonlander: Fix execution of user callback (#24475)
Ψ - Common ancestor with upstream/develop: None
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Found avr-gcc version 8.5.0
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

I have a 96-character unicode map of 64 Cyrillic characters and 32 alchemical glyphs that works perfectly except for Cyrillic А, а, Р, and р regardless of aliases under `#define`. The only common quirk between them is their unicodes:

- А: 0x0410
- Р: 0x0420
- а: 0x0430
- р: 0x0440

At first I thought it was just their ending in 0, but 0x2650 (♐) works fine. The main reason I'm using a unicode map instead of `#include "russian_keymap.h"` is because with that method `RU_YO` outputs `KC_LCBR` and `KC_RCBR`.

My map is here: [kdlrose/planck-keymaps](https://github.com/kdlrose/planck-keymaps/blob/main/keymap.c).

fauxpark, 2024-10-14:
What does the Configurator key tester log?
https://config.qmk.fm/#/test

kdlrose, 2024-10-14:
for а (0x0430):
[image]

for р (0x0440):
[image]

for А (0x0410):
[image]

for Р (0x0420):
[image]

fauxpark, 2024-10-14:
My manual input with the hex input source selected shows something slightly different:
```
KEY-DOWN - QMK: KC_LSFT Event key: Shift       Code: ShiftLeft     KeyCode: 16
KEY-DOWN - QMK: KC_LALT Event key: Alt         Code: AltLeft       KeyCode: 18
KEY-DOWN - QMK: KC_0    Event key: )           Code: Digit0        KeyCode: 48
KEY-UP   - QMK: KC_0    Event key: )           Code: Digit0        KeyCode: 48 in 67.000ms
KEY-DOWN - QMK: KC_4    Event key: $           Code: Digit4        KeyCode: 52
KEY-UP   - QMK: KC_4    Event key: $           Code: Digit4        KeyCode: 52 in 71.000ms
KEY-DOWN - QMK: KC_2    Event key: @           Code: Digit2        KeyCode: 50
KEY-UP   - QMK: KC_2    Event key: @           Code: Digit2        KeyCode: 50 in 57.000ms
KEY-DOWN - QMK: KC_0    Event key: )           Code: Digit0        KeyCode: 48
KEY-UP   - QMK: KC_0    Event key: )           Code: Digit0        KeyCode: 48 in 77.000ms
KEY-UP   - QMK: KC_LSFT Event key: Shift       Code: ShiftLeft     KeyCode: 16 in 2556.000ms
KEY-UP   - QMK: KC_LALT Event key: Alt         Code: AltLeft       KeyCode: 18 in 2413.000ms
```