qmk/qmk_firmware #24504: [Feature Request]

aceraoc, 2024-10-19:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Will QMK support GigaDevice F103x or F303x series of microcontrollers in the future?

fauxpark, 2024-10-19:
STM32F103 can be used as substitute for GD32F303, as they are mostly compatible, apparently. For example `kprepublic/bm16a/v2`. Not sure about F103.

sigprof, 2024-10-20:
The compatibility is not perfect though — in my testing of some GD32F103 and GD32F303 chips I found that the implementation of the I2C peripheral is subtly different from the original ST one, and the ChibiOS I2C driver does not work properly on those chips.

And yes, GD32F303 is more like an extended STM32F103, not STM32F303, despite its name.  And the STM32F103 series is split into low, medium, high and XL-density variants, and GD32F303 chips fall into the high-density variant due to their flash size; supporting such chips properly requires some extra defines in `board.h` to specify the proper size variant.

closed 2025-02-19 by github-actions[bot]