qmk/qmk_firmware #24519: RK61 source code

p6vdzkqs, 2024-10-24:
### Issue Description

I just got a new QMK keyboard RK61, But I have trouble to find the QMK firmware source, Has anyone used this QMK keyboard before?  

[image]

https://www.amazon.com/RK-ROYAL-KLUDGE-Mechanical-Ultra-Compact/dp/B089GN2KBT/ref=sr_1_3?crid=14P6KHEGKM9NH&dib=eyJ2IjoiMSJ9.z2wQB9GP2BYx6de4knSdid8IE25JkwbsV0e3TNXVqwh2lMX0tVMlPFSe6HTlnf_qrDNjfo7LiP027xo8jJqmH-k9hjJ8S2shwrGM7uXumYFI2-ifkzsxElPNXbcvRePaWU4Blti5pOSYpYOT-sHkRqPR1uKl6ipU3SjRW8R3CtwjQSf8l8HFCEQl-2PNUxq2JtmtcP35N1-ax-BVpyzwne4HDWeCz6R-hZwuAIGQjIw.YpcLsdpHwwSBApV-ywmwAVqX3JI7qWS1eEAclQU89kE&dib_tag=se&keywords=rk61%2Bkeyboard&qid=1729780592&sprefix=RK61%2Bke%2Caps%2C990&sr=8-3&th=1

VeyPatch, 2024-10-24:
The manufacturer doesn't share their source code. See this issue about it: https://github.com/qmk/qmk_firmware/issues/24085

filterpaper, 2024-10-24:
> It's developed based on STM32F072, just curious why it hasn't been pushed to the qmk repository yet

The manufacturer has either choosen not to make a pull request, or has yet to do so. You should reach out to them with this quotation.

p6vdzkqs, 2024-10-25:
I disassembled the keyboard and found that it's not STM32F072, it's APM32F072!

[image]

NoaHimesaka1873, 2024-11-11:
> APM32F072

Seems like "clone" of STM32 so the firmware could be legit. ("clone" because it's not exactly 1-to-1)

closed 2025-03-12 by github-actions[bot]