qmk/qmk_firmware #24530: Cannot compile with TAP_DANCE

ChrisGVE, 2024-10-26:
### Issue Description

I am getting compile errors for keymaps with a tap_dance enabled; the compiler's error messages are as follows:

`Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/eeconfig.h:24,
                 from quantum/keycode_config.h:23,
                 from quantum/quantum.h:41,
                 from ./.build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/src/default_keyboard.h:27,
                 from ./keyboards/mechlovin/adelais/keymaps/chrisgve_hhkb/keymap.c:16,
                 from quantum/keymap_introspection.c:5:
quantum/keymap_introspection.c: In function 'tap_dance_count_raw':
quantum/keymap_introspection.c:125:23: error: 'tap_dance_actions' undeclared (first use in this function); did you mean 'tap_dance_action_t'?
  125 |     return ARRAY_SIZE(tap_dance_actions);
      |                       ^~~~~~~~~~~~~~~~~
quantum/util.h:36:68: note: in definition of macro 'IS_ARRAY'
   36 | #    define IS_ARRAY(value) (!__builtin_types_compatible_p(typeof((value)), typeof(&(value)[0])))
      |                                                                    ^~~~~
quantum/keymap_introspection.c:125:12: note: in expansion of macro 'ARRAY_SIZE'
  125 |     return ARRAY_SIZE(tap_dance_actions);
      |            ^~~~~~~~~~
quantum/keymap_introspection.c:125:23: note: each undeclared identifier is reported only once for each function it appears in
  125 |     return ARRAY_SIZE(tap_dance_actions);
      |                       ^~~~~~~~~~~~~~~~~
quantum/util.h:36:68: note: in definition of macro 'IS_ARRAY'
   36 | #    define IS_ARRAY(value) (!__builtin_types_compatible_p(typeof((value)), typeof(&(value)[0])))
      |                                                                    ^~~~~
quantum/keymap_introspection.c:125:12: note: in expansion of macro 'ARRAY_SIZE'
  125 |     return ARRAY_SIZE(tap_dance_actions);
      |            ^~~~~~~~~~
quantum/util.h:47:32: error: first argument to '__builtin_choose_expr' not a constant
   47 | #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
      |                                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:125:12: note: in expansion of macro 'ARRAY_SIZE'
  125 |     return ARRAY_SIZE(tap_dance_actions);
      |            ^~~~~~~~~~
quantum/keymap_introspection.c: At top level:
quantum/keymap_introspection.c:132:27: error: 'tap_dance_actions' undeclared here (not in a function); did you mean 'tap_dance_action_t'?
  132 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                           ^~~~~~~~~~~~~~~~~
quantum/util.h:36:68: note: in definition of macro 'IS_ARRAY'
   36 | #    define IS_ARRAY(value) (!__builtin_types_compatible_p(typeof((value)), typeof(&(value)[0])))
      |                                                                    ^~~~~
quantum/keymap_introspection.c:132:16: note: in expansion of macro 'ARRAY_SIZE'
  132 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                ^~~~~~~~~~
quantum/util.h:47:32: error: first argument to '__builtin_choose_expr' not a constant
   47 | #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
      |                                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:132:16: note: in expansion of macro 'ARRAY_SIZE'
  132 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                ^~~~~~~~~~
quantum/util.h:47:31: error: expression in static assertion is not an integer
   47 | #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
      |                               ^
quantum/keymap_introspection.c:132:16: note: in expansion of macro 'ARRAY_SIZE'
  132 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                ^~~~~~~~~~
quantum/keymap_introspection.c: In function 'tap_dance_count_raw':
quantum/keymap_introspection.c:126:1: error: control reaches end of non-void function [-Werror=return-type]
  126 | }
      | ^
quantum/keymap_introspection.c: In function 'tap_dance_get_raw':
quantum/keymap_introspection.c:139:1: error: control reaches end of non-void function [-Werror=return-type]
  139 | }
      | ^
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [.build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o] Error 1
make: *** [mechlovin/adelais/standard_led/arm/rev4/apm32f103:chrisgve_hhkb] Error 1
Make finished with errors
`

teleyinex, 2024-10-26:
I'm having the same problem. I have updated to the latest version in main branch but I still have the same issue.

waffle87, 2024-10-26:
The `tap_dance_action_t tap_dance_actions` array needs to be in your `keymap.c` file, or `INTROSPECTION_KEYMAP_C` flag set appropriately in your `rules.mk` file.

teleyinex, 2024-10-26:
Oh, thanks a lot for the information. Trying that now.

teleyinex, 2024-10-26:
I can confirm the fix was that and now I'm a really happy user with tap dance on my keyboard. @ChrisGVE you can take a look into my code to see how I have enabled it:  https://github.com/teleyinex/qmk_userspace/blob/main/keyboards/crkbd/keymaps/teleyinex/keymap.c#L39

ChrisGVE, 2024-10-26:
Thanks. Unfortunately, the code for the tap_dance is in my userspace (activated only if `TAP_DANCE` is enabled for the layout), as I use the definition for multiple keyboards.

I've tried to add `INTROSPECTION_KEYMAP_C = chrisgve.c` in the user space `rules.mk`, and once in the keyboard layout `rules.mk`.

With either variant, the compilation is successful, but the binary build fails, complaining of multiple definitions from my userspace code. It is progress, but I'm not sure how to proceed from here.

```
Linking: .build/mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb.elf                 [ERRORS]
 |
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `cps_ctl_reset'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `caps_lock'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `cur_dance'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `cps_ctl_finished'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `reset_rgb'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `set_rgb'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `set_nav_1_rgb'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `set_nav_2_rgb'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `set_caps_rgb'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `set_adj_rgb'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
[... cut ...]
 | /Applications/ArmGNUToolchain/13.3.rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.3.1/../../../../arm-none-eabi/bin/ld: .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/quantum/keymap_introspection.o (symbol from plugin): in function `cps_ctl_reset':
 | (.text+0x0): multiple definition of `tap_dance_actions'; .build/obj_mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb/chrisgve.o (symbol from plugin):(.text+0x0): first defined here
 | collect2: error: ld returned 1 exit status
 |
gmake: *** [builddefs/common_rules.mk:280: .build/mechlovin_adelais_standard_led_arm_rev4_apm32f103_chrisgve_hhkb.elf] Error 1
```

ChrisGVE, 2024-10-26:
Update: I have removed `SRC =+ chrisgve.c` from `rules.mk` and added `INTROSPECTION_KEYMAP_C = chrisgve.c` to my userspace `rules.mk`.
In this case it successfully builds the firmware and it seems to be working properly on the keyboard.
I am however unsure whether this approach is proper or if I should do it differently.

waffle87, 2024-10-28:
> I have removed `SRC =+ chrisgve.c` from `rules.mk` and added `INTROSPECTION_KEYMAP_C = chrisgve.c` to my userspace `rules.mk`.

This is the correct approach. `INTROSPECTION_KEYMAP_C` includes the file's code:
https://github.com/qmk/qmk_firmware/blob/7c9e650b1c8ba6b1fd2bfffa8dc64e27504e7e37/quantum/keymap_introspection.c#L8-L10

ChrisGVE, 2024-10-28:
Great thanks a lot for the support

closed 2024-10-28 by ChrisGVE

chkpwd, 2024-10-29:
I followed the foundings listed by @ChrisGVE, however, the keymap.c does not recognize the enum set for the tap dance configuration. Would kindly have a look?

https://github.com/chkpwd/qmk-configs/pull/2

adaviloper, 2024-11-10:
How do I leverage `INTROSPECTION_KEYMAP_C` for including my `combos.c` file in my userland directory and _also_ my `tap_dance.c` file that is also in my userland directory?

chkpwd, 2024-11-11:
You can use my repo as an example. @adaviloper https://github.com/chkpwd/qmk-configs/blob/main/users/chkpwd/rules.mk